OpenBCI/OpenBCI_GUI #6: Firmware V2 Updates / Additional GUI Interaction

conorrussomanno, 2016-06-29:
[ ] Get channel # — tells you channel number of your host/device pair
[ ] Set channel # — sets host and device, if device present. In case of failure to change, reverts back to original
[ ] set channel override — only sets the host... use this in case of not knowing device channel # ...
[ ] set channel override LOOP — loop host through channels until it makes contact with device
[ ] Set poll time — in case you're running into programming issues / code to radios (Serial-related timing)... FAST / MEDIUM / SLOW / OTHER ___
[ ] time syncing (starting, parsing, changing packet types, etc.) ... this will need to be worked into Serial code ... 
[ ] Parsing soft reset message for firmware v2 — GUI needs to work with V1 & V2 firmware
[ ] Remove sending delays between commands — firmware V2 can stack ASCII commands

pushtheworldllc, 2016-06-29:
Here are the new commands
```
#define OPENBCI_HOST_TIME_SYNC '<'
#define OPENBCI_HOST_TIME_SYNC_ACK ','
#define OPENBCI_HOST_CHANNEL_GET 0xF0
#define OPENBCI_HOST_CHANNEL_SET 0xF1
#define OPENBCI_HOST_CHANNEL_SET_OVERIDE 0xF2
#define OPENBCI_HOST_POLL_TIME_GET 0xF3 // TODO: Implement
#define OPENBCI_HOST_POLL_TIME_SET 0xF4
```

cfausn, 2016-11-21:
@aj-ptw Check out my post in the docs repo: https://github.com/OpenBCI/Docs

Is this tutorial sufficient? Anything I should change or add to the widget itself?

closed 2017-01-19 by aj-ptw