OpenBCI/OpenBCI_GUI #22: splitLineCSV throws error

pushtheworldllc, 2016-08-10:
Trying to run master and getting this error...

[image]

pushtheworldllc, 2016-08-10:
This is because of a bug with processing 3.1.2, do not update!

pushtheworldllc, 2016-08-10:
From the processing guys, we should never have used `splitLineCSV()` and functions like these are subject to change in minor updates to processing!! See this issue here! https://github.com/processing/processing/issues/4611

cfausn, 2016-09-02:
I've found that passing the buffer to splitLineCSV() fixes this bug with newer processing. Simply change it to splitLineCSV(line, reader)

aj-ptw, 2016-09-02:
I disagree. `splitLineCSV` is a protected function and should not be used in the application.

cfausn, 2016-09-02:
haha interesting. For the newest version of processing though it doesn't let you compile without passing in the reader, dunno why.

aj-ptw, 2016-09-02:
They changed the interface for that function from one arg to two args... see #25 PR for a fix https://github.com/aj-ptw/OpenBCI_GUI_v2.0/commit/53776fddab94a02700cbcc1cd8dc9e107bd7e33b#diff-5fbebc9acb054941aa8e1303d08be960R241

closed 2016-11-02 by aj-ptw