OpenBCI/OpenBCI_GUI #29: OpenBCI_GUI Freezes on macOS Sierra with LIVE Data Source

valencra, 2016-10-08:
I just received my OpenBCI 32-bit board, and, when I ran it with the GUI, it just freezes instead of establishing a connection:

[image]

aj-ptw, 2016-10-08:
You hit start and just nothing? Is there any print outs you see on the bottom of the window? Have you tried the older version one build that can be downloaded from the website.

valencra, 2016-10-08:
Yes after making all the required selections and pressing start system it freezes. In the tutorial it says that the bottom status should change to “Initializing communication w/ your OpenBCI board”, but that does not happen. If I wait 5 seconds, which is the amount of time indicated in the tutorial, nothing happens and the screen becomes blank. If I try to maximize the window, it completely exits the program.

aj-ptw, 2016-10-08:
Could be. Maybe try downloading the repo for the GUI and run it from processing.

valencra, 2016-10-09:
I tried downloading the repo and running it from processing, and it still has the same problem:
[image]

The snapshot does not show it, but there's just a spinner that never goes away

aj-ptw, 2016-10-09:
Does synthetic work?

valencra, 2016-10-09:
On the standalone app, yes:
[image]

When run with processing, it freezes as well:
[image]

aj-ptw, 2016-11-02:
Closing for now

closed 2016-11-02 by aj-ptw

valencra, 2016-11-02:
Oh is there no possible solution to the problem? Please let me know if you plan on opening it once again in the future.

aj-ptw, 2016-11-02:
I closed because this is still alpha and there is a big chance your issue was/will be solved by the time this is released by it go live in december

valencra, 2016-11-02:
I see. Thank you for letting me know, AJ.