OpenBCI/OpenBCI_GUI #38: Upgrade Playback Functionality

aj-ptw, 2016-11-02:
New features for playback

- [ ]

retiutut, 2018-08-03:
@andrewjaykeller This issue has been resolved by developing a **functional playback slider sub-widget** that appears during playback. A branch has been made off the development branch:
https://github.com/retiutut/OpenBCI_GUI/tree/playback-ts-scrubbing

retiutut, 2018-08-04:
[image]

retiutut, 2018-08-04:
[image]

daniellasry, 2019-01-05:
It looks like this feature has been implemented successfully. Verified in 4.0.3 
Feel free to open separate issues for any enhancements regarding the playback slider.

closed 2019-01-05 by daniellasry