OpenBCI/OpenBCI_GUI #48: Enh: Open/Close Data Recording File in GUI

aj-ptw, 2016-11-02:
Be able to open/close or stop/start an output file from within the GUI without having to disconnect and got to the control panel.

retiutut, 2018-08-05:
This issue is being addressed in https://github.com/retiutut/OpenBCI_GUI/tree/playback-widget !

Created a new playback widget to solve this issue, potentially #55, and to add future playback capabilities.

Currently functional using a second dialog box fitted inside of the new widget. Only appears when using playback mode.

retiutut, 2018-08-17:
- [x] __When loading playback from control panel, # chan auto-sets!__ This occurs before the system is initialized. 
- [ ] Next step is to auto-set number of channels being loading from within GUI, and have the system adapt to the change in number of channels if necessary.

daniellasry, 2019-01-05:
This is a duplicate of #400

retiutut, 2019-01-05:
@daniellasry This is not a duplicate. This issue should be closed since the playback widget has been merged to master as of v4.0.0

daniellasry, 2019-01-05:
Thanks for clarifying. I'll close this one.

closed 2019-01-05 by daniellasry