OpenBCI/OpenBCI_GUI #118: GUI freezes at start on Mac OS Sierra 10.12.3

compmonks, 2017-02-02:
Hi all,
the gui freezes when system starts.
connection with the dongle and the cyton is up.

16channels cyton +daisy
GUI mac version 2.0.0
OS Mac sierra 10.12.3
ftdi driver 2.3

should I downgrade my OS or is there a quick fix?
cheers

aj-ptw, 2017-02-02:
Where are you launching the Application from? What directory is it in?

compmonks, 2017-02-02:
from the mac app directory

aj-ptw, 2017-02-02:
Can you please screen shot

compmonks, 2017-02-02:
[image]

compmonks, 2017-02-02:
[image]

aj-ptw, 2017-02-02:
Sorry I meant of your application folder, I want to see if the OpenBCI GUI app is within another folder in your application for or not.

compmonks, 2017-02-02:
here you go.
I have also tried with the application outside of the openbci folder but the results are similar

[image]

aj-ptw, 2017-02-02:
> outside of the openbci folder

Like this?
[image]

compmonks, 2017-02-02:
yes.
I finally got it working with the application outside the openbci folder but with keeping the saveddata folder inside (see below).
[image]

aj-ptw, 2017-02-02:
@compmonks could you please elaborate more? 

Did you see this line:
http://docs.openbci.com/OpenBCI%20Software/01-OpenBCI_GUI#the-openbci-gui-using-the-openbci-gui-as-a-standalone-application-install-the-application

specifically: _**Note:** for Mac OS Sierra, you must put the application in your /Applications folder in order for it to work properly._

closed 2017-02-02 by aj-ptw

compmonks, 2017-02-02:
@aj-ptw thank you
yes I saw the previous indications in the tutorial.
My mistake was to put both the application and the "SavedData" folder directly in the applications directory. In my case it seems that it works only if the "SavedData" folder stays inside the "OpenBCI_GUI_200_MACOSX" folder.