OpenBCI/OpenBCI_GUI #119: Add timezone info or Unix/Iso time in milliseconds to raw data output

prefrontalvortex, 2017-02-04:
I'm developing an ERP (event related potential) experiment program to use in conjunction with openBCI and [ERPLAB](https://github.com/lucklab/erplab). I need to synchronize down to the millisecond the events that occur in the ERP program with the EEG output data. Currently, the data is output as HH:MM:SS.mms, which gives me the ms resolution, but is timezone-unaware, so in order to line it up with the output of the ERP program, I have to make assumptions about timezone. This may affect portability down the line.

Unix time provides an unambiguous, UTC-based timestamp which is parseable by any datetime library. Isotime (ISO 8601) is also unambiguous and human-readable.

It would be nice if there were an additional column with Unix time in ms, e.g. 1484769231000 or 1484769231.000, or similar, or if there were an option to set the timestamp format. Making it the default would be rather nice, but I don't want to rustle any jimmies by implying this is the One True Time format. 

I've hacked this feature into my own branch, but it would be nice if something made its way into the release package. 

Thanks!

aj-ptw, 2017-02-16:
@prefrontalvortex hey thanks for the feature request! Will pass this on to @conorrussomanno and @biomurph for something to add

daniellasry, 2019-01-05:
Closing. This looks like it was fixed/implemented. In GUI version 4.0.3 the raw data files have a regular timestamp, as well as a UNIX time stamp.

closed 2019-01-05 by daniellasry