OpenBCI/OpenBCI_GUI #120: V2 GUI freezes playing back recorded .txt / .edf incompatible

topant, 2017-02-05:
Hi,

I use OpenBCI GUI v2.1 x64 and Ultracortex 8-channel Mark III on Win7 x64.
Record EEG signals and save it locally works fine but when I try to playback signals from file, OpenBCI GUI freeze.
If I try to open file (.txt or .edf) on OpenBCI GUI v1, everything works fine.

Furthermore, I tried to open .edf file in EEGLAB 13.6.5b (Matlab plugin) and I got the following message:
---------------------------------------------------------------------------------------------------
EEGLAB error in function edfannot2evt() at line 40:
Attempted to access ix(1); index out of bounds because numel(ix)=0.
---------------------------------------------------------------------------------------------------

I suppose that the problem is in the OpenBCI GUI .edf file because all other .edf files recorded with other EEG devices and applications, I can open without any problem.

Any ideas what is wrong?

topant, 2017-02-05:
[OpenBCI-RAW-2017-02-04_22-33-00.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/753217/OpenBCI-RAW-2017-02-04_22-33-00.txt)

topant, 2017-02-05:
[OpenBCI-RAW-2017-02-04_22-33-00 - ORG.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/753220/OpenBCI-RAW-2017-02-04_22-33-00.-.ORG.txt)

aj-ptw, 2017-02-05:
Hi @topant thanks for the reports!

I moved the `.edf` over to #121 and tagged you in it.

Totally seeing this error on my end too! I'll patch it and let you know here when it's fixed!

topant, 2017-02-06:
All right, thanks for info.

aj-ptw, 2017-02-07:
working on this right now

aj-ptw, 2017-02-07:
Looks like the problem is [in ControlPanel.pde](https://github.com/OpenBCI/OpenBCI_GUI/blob/master/OpenBCI_GUI/ControlPanel.pde#L152)

```java
...
else if(newDataSource == DATASOURCE_PLAYBACKFILE){
      updateToNChan(8);
    } 
...
```

aj-ptw, 2017-02-07:
@topant the problem was solved on the [original GUI here](https://github.com/OpenBCI/OpenBCI_Processing/pull/49)

aj-ptw, 2017-02-07:
#124 fixes this issue. it's in development and will be merged soon. will close this issue when merged with master. for now it's in the development branch!

topant, 2017-02-13:
Hi,

when will you release a new version of OpenBCI GUI with these fixes?

closed 2017-02-16 by aj-ptw

aj-ptw, 2017-02-16:
Hi running out of time for this week but was able to get a pre-release up https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v2.1.2 so please try that and let me know!

aj-ptw, 2017-03-16:
@topant the new GUI has been released, please download the new gun from either openbci.com

PR #124 (2017-02-07): Fixed locale to US always for ODF format - #120
### Bug Fixes
* Fix #120 - Locale dependent formatting cause issue with output ODF file.
PR #131 (2017-02-16): Development
# 2.1.2
 
### Bug Fixes
* Fix #120 - Locale dependent formatting cause issue with output ODF file.