OpenBCI/OpenBCI_GUI #121: Bug: .edf incompatible

aj-ptw, 2017-02-05:
Hi,

I use OpenBCI GUI v2.1 x64 and Ultracortex 8-channel Mark III on Win7 x64.
Record EEG signals and save it locally works fine but when I try to playback signals from file, OpenBCI GUI freeze.

I tried to open .edf file in EEGLAB 13.6.5b (Matlab plugin) and I got the following message:

```matlab
EEGLAB error in function edfannot2evt() at line 40:
Attempted to access ix(1); index out of bounds because numel(ix)=0.
```

I suppose that the problem is in the OpenBCI GUI .edf file because all other .edf files recorded with other EEG devices and applications, I can open without any problem.

Any ideas what is wrong?

(taken from #120 @topant )

aj-ptw, 2017-02-05:
@topant can you please upload the EDF file. Sorry about this! Will try recording from 16 chan when i get in tomorrow.

topant, 2017-02-06:
Hi,

I get error when I try to upload EDF file:

> We don’t support that file type. with a PNG, GIF, JPG, DOCX, PPTX, XLSX, TXT, PDF, or ZIP. 

so I'll send you on the mail.

topant, 2017-02-06:
Hi,

please find EDF files in attachment regarding to https://github.com/OpenBCI/OpenBCI_GUI/issues/121.

Regards,

Ante

aj-ptw, 2017-06-27:
@topant can you please checkout the master and record some new data and let me know if it is working now?

topant, 2017-07-19:
@aj-ptw Sorry for late answer.
It works fine now.
Thanks a lot!

closed 2017-07-19 by aj-ptw