OpenBCI/OpenBCI_GUI #123: Visualizers not working for Ganglion

matthew-groth, 2017-02-07:
I learned how to use the Processing version and looked through the code. Now I understand that visualizers are purposefully turned off for the Ganglion. I turned them on to see what would happen, and they did not work. I'm guessing that this is because OpenBCI GUI doesn't officially support Ganglion yet? Or is there a way I could get visualizers to work?

aj-ptw, 2017-02-07:
@matthew-groth please upload screen shots. I'm not sure which visualizers you are talking about.

matthew-groth, 2017-02-07:
Here is what my GUI looks like after initialization:

[image]

Here is the output, where the last relevant line is "OpenBCI_GUI: initSystem: -- Init 4 -- 14159"

[image]

OpenBCI_GUI does not go into POSTINIT mode if datasource is Ganglion:
`      if (eegDataSource != DATASOURCE_GANGLION && eegDataSource != DATASOURCE_NORMAL_W_AUX) {
        systemMode = SYSTEMMODE_POSTINIT; //tell system it's ok to leave control panel and start interfacing GUI
      }`

So I removed the ganglion check so I could see the graphs. But even after starting data stream, nothing happens:

[image]

And the last relevant printed output in case it is useful: 

[image]

aj-ptw, 2017-02-07:
Hey there is some connection issue going on here because you can see the message:
```bash
OpenBCI_Ganglion: parseMessage: default: k,400,;
```
because if you checkout the HUB docs https://github.com/OpenBCI/Docs/blob/master/OpenBCI%20Software/07-OpenBCI_Hub.md#command it's a standard error response code. It means your not able to connect to the Ganglion. That's why there is a freeze.

Please post the output printed **before** that last screen shot

matthew-groth, 2017-02-07:
Solved it! Thanks to pushtheworld on [my forum post](http://openbci.com/forum/index.php?p=/discussion/comment/5522#Comment_5522) for telling me all I had to do was remove the dongle.

I got signal!
[image]

I am going to make a couple of pushes to help other Mac users avoid this issue.

aj-ptw, 2017-02-07:
@pushtheworld saves the day! woo ;)

closed 2017-02-07 by aj-ptw