OpenBCI/OpenBCI_GUI #125: Bug: Unable to Verify Device Status In GUI

dillonhuffman, 2017-02-09:
Hello all,
I recently started working with the Cyton 32-bit board and have been trying to get it up and running.  After installing the latest GUI and FTDI drivers, I tried to run the GUI. If I plug in the dongle, power on the board, run the GUI, and start data stream, everything seems to function correctly. However, if I try to check device status in the GUI (select COM>STATUS), I get the following message: 
"Error reading from serial port. Try a different port?"

If I try to start the stream after this, it does not work, and I have to close and re run the GUI to stream data.

Any suggestions? Should I be concered about this?

wjcroft, 2017-02-09:
Relevant Forum thread,

http://openbci.com/forum/index.php?p=/discussion/950/data-streaming-but-cannot-verify-connection-status

aj-ptw, 2017-02-13:
V1 firmware does not have the status features, they were introduced in V2.

dillonhuffman, 2017-02-13:
Ah, I assumed that since we bought the board after the newest firmware release date that it would have the new version.  I will update it! Thanks!

closed 2017-02-16 by aj-ptw