OpenBCI/OpenBCI_GUI #126: High CPU usage

dado93, 2017-02-09:
Hi,

I'm experiencing very high CPU usage with the OpenBCI GUI v2 (and also with the custom software that I developed in Processing to interface with the OpenBCI boards): the %CPU reaches on average 100%, with peaks well above it (see attached figure). 

My hardware specs are:

- OpenBCI 32 - bit with custom v2 firmware (the only difference with the default firmware is the analog reading from an analog PIN)

- OpenBCI Ganglion Board 

- Processing 3.2.1

- Mac OS 10.11.6

[image]

aj-ptw, 2017-02-09:
Try changing the layout and not using the headplot

dado93, 2017-02-09:
I changed the layout: with the time series only, it is about 50% when not streaming, 90% when streaming.

[image]
[image]

dado93, 2017-02-10:
Here is the hardware info of my computer:
  Model:	MacBook Pro
  Processor:	Intel Core i7 dual core 3.1 GHz
  Velocità processore:	3,1 GHz
  RAM:	8 GB

aj-ptw, 2017-02-16:
@dado93 hi there, sorry for the delayed response. 
This is most likely due to this line https://github.com/OpenBCI/OpenBCI_GUI/blob/master/OpenBCI_GUI/OpenBCI_GUI.pde#L122 or 
```java
final int nPointsPerUpdate = 10;
```

Which will cause performance issues. Lowering this number results in less CPU usage.

aj-ptw, 2017-08-26:
Hi this is fixed in the upcoming v3 GUI with variable frame rates and not loading the head plot by default. also we introduced multiprocessing to do the data collection in a whole different process.

closed 2017-08-26 by aj-ptw