OpenBCI/OpenBCI_GUI #128: BLE on Mac

HMBTECP, 2017-02-14:
i have problems connecting the OpenBCI Ganglion to the GUI.
the Bluetooth is activated and yet the the application give me code 412Error Turn Bluetooth on.
and i can't find the board.
and if i use a csr bluetooth dongle i can find the board but i can't start the system.

aj-ptw, 2017-02-15:
What OS?

HMBTECP, 2017-02-16:
MacOS Sierra
10.12.2

aj-ptw, 2017-02-16:
@HMBTECP this sounds super frustrating, Sierra has been causing headaches around the world with their new permissions and such. Would you mind reviewing the [Ganglion Getting Started Guide](http://docs.openbci.com/Tutorials/02-Ganglion_Getting%20Started_Guide#ganglion-getting-started-guide-downloadrun-the-openbci-gui-on-macos) in the tutorials? With Sierra, the location of the GUI application is extremely important, see this excerpt from the guide:

> NOTE: If you are using a Mac with macOS Sierra, you MUST move the OpenBCI_GUI.app into your Applications folder. You also need to change your Security & Privacy settings to allow apps from Anywhere. Go to the end of this page for instructions on how to do that, then come back to conintue this tutorial.

That being said, here is a screen shot of what my finder looks like with the GUI:
[image]

You can see the executable app placed right in the applications folder, this is new to Sierra.

Also please do not use the Dongle with Mac OS, just use the built in BLE.

HMBTECP, 2017-02-17:
@aj-ptw 
Thank you very much for your detailed answer 
but i already did all these steps.
and i already read about the getting started in Ganglion webpage

aj-ptw, 2017-05-18:
Is this issue still persisting?

ds1, 2017-07-02:
This problem still persists for me. I've tried both as a standalone application and running from the .pde file.

-Updated FTDI drivers
-Ganglion Hub is running

I'm running Mac Sierra 10.12.3 and have followed the instructions here: http://docs.openbci.com/OpenBCI%20Software/00-OpenBCISoftware

I've tried opening Bluetooth Preferences and no devices appear.
They don't appear when I refresh the list in the Open BCI GUI either in the LIVE (from Ganglion) option.

aj-ptw, 2017-07-06:
Are you using a BLE dongle? Mac does not use dongle

armengol, 2017-08-04:
I'm running Mac Sierra 10.12.5. I just bought a Ganglion board and I am facing the same problem. My board does not appear in the list of BLE devices in OpenBCI_GUI. I followed all the steps in "Ganglion Getting Started Guide" but problem is persisting. I can find the device using a BLE device scanner app, so the problem does not come from the board itself.
Any suggestions?

Thank you.

WillWillyto, 2017-08-08:
Yup, same problem here. Running macOS Sierra Version 10.12.5 and neither the GUI nor the built in Bluetooth system are detecting my Ganglion board. Moved the app to the Applications folder as mentioned above and gave the administrator rights to the app but still nothing happens.
Suggestions are very welcome.

Thank you.

aj-ptw, 2017-08-08:
Can you please try running an earlier version such as v2.2.0

http://openbci.com/index.php/software

WillWillyto, 2017-08-08:
worked. love you!

aj-ptw, 2017-08-08:
Weird! Looking into the problem.

armengol, 2017-08-08:
It worked thanks!!

aj-ptw, 2017-08-08:
Did you recently update? I know my computer has been asking me for a while

closed 2017-08-22 by aj-ptw

VessoVit, 2017-10-10:
10.13 same issue

aj-ptw, 2017-10-10:
@VessoVit please open a new issue with the GUI version you're running and os?

rschaef1, 2018-11-15:
I am having the same issue with OS Mojave 10.14, and unfortunately neither version 3.2 or 2.2 has worked for me. When I updated my OS the ganglion board just stopped showing up on bluetooth