OpenBCI/OpenBCI_GUI #129: Wrong time stamp

Jianbin-IBM, 2017-02-15:
Hi, below is the raw data from openbci, it says sample rate is 250Hz. The last column is timestamp. It increases like 883,884,884,884,885. But I expect it would be kind of 883,887,891(with 0.004 step because 250Hz sampling) May I know how to solve it? Thank you very much!

%OpenBCI Raw EEG Data
%
%Sample Rate = 250.0 Hz
%First Column = SampleIndex
%Last Column = Timestamp 
%Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
0, 75561.05, 64046.38, -187500.02, -187500.02, 46287.06, 46024.03, 46088.63, 64357.91, 0.016, 0.474, 0.114, 12:30:54.403
1, 75540.76, 64015.62, -187500.02, -187500.02, 46587.18, 46188.85, 46114.80, 64292.18, 0.000, 0.000, 0.000, 12:30:54.883
2, 75085.32, 63538.88, -187500.02, -187500.02, 46615.46, 46460.29, 45689.04, 64611.76, 0.000, 0.000, 0.000, 12:30:54.884
3, 74793.98, 63247.59, -187500.02, -187500.02, 46317.13, 46468.63, 45363.09, 64885.01, 0.000, 0.000, 0.000, 12:30:54.884
4, 75086.05, 63556.58, -187500.02, -187500.02, 46099.87, 46181.70, 45600.33, 64711.41, 0.000, 0.000, 0.000, 12:30:54.884
5, 75568.36, 64039.73, -187500.02, -187500.02, 46286.95, 46017.30, 46072.11, 64330.53, 0.000, 0.000, 0.000, 12:30:54.885
6, 75551.42, 64017.56, -187500.02, -187500.02, 46598.31, 46184.67, 46105.88, 64257.84, 0.000, 0.000, 0.000, 12:30:54.885
7, 75091.38, 63544.78, -187500.02, -187500.02, 46615.19, 46453.54, 45664.03, 64578.17, 0.028, 0.936, 0.228, 12:30:54.885
8, 74852.21, 63312.16, -187500.02, -187500.02, 46351.35, 46482.86, 45386.29, 64852.54, 0.000, 0.000, 0.000, 12:30:54.886
9, 75167.91, 63648.67, -187500.02, -187500.02, 46168.55, 46229.98, 45658.82, 64698.06, 0.000, 0.000, 0.000, 12:30:54.886
10, 75657.39, 64148.92, -187500.02, -187500.02, 46366.32, 46085.83, 46147.37, 64334.53, 0.000, 0.000, 0.000, 12:30:54.886

aj-ptw, 2017-02-15:
Looks like you need to lower your serial port latency, which type of computer are you on?

Jianbin-IBM, 2017-02-16:
Hi,

I am using  Windows 7, 64bit, Lenovo W530.

Jianbin Tang,
Research Staff Member
IBM Research Australia at Melbourne
TEL:   61-399213860
FAX:   61-399213890
e-mail: jbtang@au1.ibm.com

            <author@noreply.github.com>
Date:	16/02/2017 09:55 AM

aj-ptw, 2017-02-16:
@Jianbin-IBM could you please try lowering your serial port latency to `1`ms and your Transmit/Receive buffer both to `64`bytes?

Please see instructions here: http://openbci.com/forum/index.php?p=/discussion/200/latency-timer-windows-control-panel-setting

Jianbin-IBM, 2017-02-16:
Hi,

After setting the latency to 1ms, the timestamp is becoming more
reasonable. Most of them increase 0.004s, with some of them 0.003 or
0.005m. And there are even some gap is 0.007 which is too big. I am not
sure the reason here. The time stamp is from cyton board itself or
computer? If it is from cyton board, should be very equal.

Here is a piece of the raw data:

18, 74250.59, 106135.87, 6282.31, -56839.99, -119159.25, -358.03,
-32441.68, -27209.47, 0.000, 0.000, 0.000, 09:25:28.201
19, 74313.58, 106208.13, 6354.00, -57405.96, -119629.69, -805.62,
-32373.30, -27132.56, 0.000, 0.000, 0.000, 09:25:28.204
20, 74188.92, 106109.41, 6238.17, -57175.05, -119492.71, -595.58,
-32317.36, -27269.87, 0.000, 0.000, 0.000, 09:25:28.207
21, 74065.38, 105977.30, 6098.90, -56577.54, -118974.13, -97.54, -32360.88,
-27412.87, 0.000, 0.000, 0.000, 09:25:28.212
22, 74089.05, 105992.21, 6121.87, -56368.55, -118728.69, 55.86, -32435.80,
-27373.80, 0.000, 0.000, 0.000, 09:25:28.216
23, 74212.33, 106090.72, 6237.57, -56875.91, -119038.73, -372.09,
-32453.53, -27245.26, 0.000, 0.000, 0.000, 09:25:28.220
24, 74301.20, 106181.73, 6315.06, -57451.60, -119492.58, -827.62,
-32395.54, -27157.30, 0.092, 0.950, 0.196, 09:25:28.224
25, 74184.05, 106110.84, 6230.79, -57221.74, -119363.23, -620.24,
-32328.15, -27268.43, 0.000, 0.000, 0.000, 09:25:28.229
26, 74063.87, 105975.83, 6104.02, -56614.82, -118845.02, -127.43,
-32369.66, -27405.61, 0.000, 0.000, 0.000, 09:25:28.232
27, 74072.98, 105998.54, 6116.93, -56392.02, -118597.13, 26.80, -32443.31,
-27379.93, 0.000, 0.000, 0.000, 09:25:28.235
28, 74248.96, 106124.00, 6250.80, -56892.61, -118910.31, -398.71,
-32473.71, -27226.86, 0.000, 0.000, 0.000, 09:25:28.240
29, 74342.30, 106212.24, 6335.45, -57446.71, -119388.28, -843.85,
-32387.30, -27138.01, 0.000, 0.000, 0.000, 09:25:28.244
30, 74202.89, 106095.19, 6213.81, -57207.32, -119241.10, -619.84,
-32318.63, -27272.97, 0.000, 0.000, 0.000, 09:25:28.251
31, 74080.56, 105987.74, 6088.77, -56616.12, -118730.27, -118.82,
-32367.36, -27390.86, 0.000, 0.000, 0.000, 09:25:28.254

Jianbin Tang,
Research Staff Member
IBM Research Australia at Melbourne
TEL:   61-399213860
FAX:   61-399213890
e-mail: jbtang@au1.ibm.com

            <mention@noreply.github.com>
Date:	17/02/2017 04:56 AM

http://openbci.com/forum/index.php?p=/discussion/200/latency-timer-windows-control-panel-setting

aj-ptw, 2017-02-16:
The time stamp you are seeing is from the PC, there is a way to get time stamps from the board, however it's not been implemented in the GUI yet. Wireless time syncing is only in the OpenBCI_NodeJS repo, if you want to write the time sync/stamp code for this repo. That being said, we would love a PR with this feature for this repo!

Jianbin-IBM, 2017-02-16:
Hi, Aj: thanks for your quick response! Here is the updated data after I change buffer size to be 64. It is actually similar as 4096 buffer size: not always 0.004. May I know if it is already equally sampled data in this case? If so, I will not care about the different gap of the timestamp for now. And next step I will try to use python interface to get the data. Hope there is a way to get time stamp from board with python interface.
14, -182949.31, -182936.27, -183002.06, -183040.63, -175749.02, -176877.75, -183092.00, -182999.02, 0.000, 0.000, 0.000, 09:45:59.463
15, -182950.75, -182936.39, -183003.25, -183041.47, -175779.75, -176728.33, -183092.86, -182999.83, 0.012, 0.926, 0.316, 09:45:59.467
16, -182950.94, -182933.98, -183004.23, -183041.73, -175858.67, -177348.88, -183093.73, -183001.19, 0.000, 0.000, 0.000, 09:45:59.472
17, -182950.09, -182932.84, -183003.81, -183041.16, -175862.61, -177869.67, -183093.61, -183001.42, 0.000, 0.000, 0.000, 09:45:59.476
18, -182950.36, -182935.44, -183003.19, -183041.27, -175810.66, -177536.34, -183093.39, -183000.67, 0.000, 0.000, 0.000, 09:45:59.479
19, -182951.03, -182938.00, -183003.78, -183042.31, -175751.00, -176881.64, -183093.89, -183000.78, 0.000, 0.000, 0.000, 09:45:59.483
20, -182953.77, -182939.39, -183006.27, -183044.28, -175782.69, -176737.56, -183095.83, -183002.75, 0.000, 0.000, 0.000, 09:45:59.487
21, -182956.58, -182939.59, -183009.84, -183047.28, -175864.02, -177362.73, -183099.45, -183006.84, 0.000, 0.000, 0.000, 09:45:59.491
22, -182953.94, -182936.75, -183007.61, -183045.17, -175866.11, -177880.78, -183097.73, -183005.42, 0.000, 0.000, 0.000, 09:45:59.495
23, -182954.41, -182939.52, -183007.30, -183045.33, -175814.05, -177542.19, -183097.42, -183004.92, 0.000, 0.000, 0.000, 09:45:59.499
24, -182957.30, -182944.27, -183010.11, -183048.48, -175772.22, -176882.13, -183099.92, -183006.95, 0.000, 0.000, 0.000, 09:45:59.504
25, -182958.72, -182944.31, -183011.23, -183049.31, -175790.78, -176738.14, -183100.83, -183007.78, 0.018, 0.924, 0.310, 09:45:59.507
26, -182959.08, -182942.11, -183012.39, -183049.89, -175870.14, -177361.25, -183102.00, -183009.44, 0.000, 0.000, 0.000, 09:45:59.511
27, -182959.02, -182941.80, -183012.75, -183050.16, -175872.95, -177870.97, -183102.72, -183010.38, 0.000, 0.000, 0.000, 09:45:59.515
28, -182960.67, -182945.72, -183013.48, -183051.39, -175819.53, -177528.09, -183103.67, -183010.98, 0.000, 0.000, 0.000, 09:45:59.519
29, -182960.00, -182946.77, -183012.72, -183051.11, -175759.77, -176873.28, -183102.78, -183009.58, 0.000, 0.000, 0.000, 09:45:59.523

aj-ptw, 2017-02-16:
From the GUI you use UDP or LSL.

For a more stable, robust, lightweight, but without a user interface you should use the  [NodeJS Python example](https://github.com/OpenBCI/OpenBCI_NodeJS/tree/master/examples/python) in the NodeJS repo. There is a PR for an LSL example using the same framework, plus if you have firmware V2, NTP inspired time syncing is baked into the NodeJS side and all you use the Python to do real time Analytics.

closed 2017-06-27 by aj-ptw