OpenBCI/OpenBCI_GUI #130: Connection refused error on startTCPClient on Mac

matthew-groth, 2017-02-16:
I have my ganglion here in between 2 macs, the one I regularly use ( Macbook Pro) and another one I am using temporarily to test the Ganglion (an iMac).

The ganglion is on, and I am not using the USB dongle on either mac. Bluetooth is on for both macs.

**On my regular computer (macbook pro), OpenBCI GUI works fine. On the iMac, there is this error.**

Just to be sure:

- I definitely put the ganglion hub into the data folder on the imac
- I definitely installed FTDIUSBSerialDriver_v2_2_18.dmg on the imac
- I definitely put all the libraries in the right folder on the iMac
- The code is unaltered, and the tcpGanglionPort and tcpGanglionIP are both still the default values and the same as on my macbook pro.

The error occurs here:
`java.net.ConnectException: Connection refused (Connection refused)
	at java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at java.net.Socket.connect(Socket.java:538)
	at java.net.Socket.<init>(Socket.java:434)
	at java.net.Socket.<init>(Socket.java:211)
	at processing.net.Client.<init>(Unknown Source)
	at OpenBCI_GUI$OpenBCI_Ganglion.startTCPClient(OpenBCI_GUI.java:8480)
	at OpenBCI_GUI$OpenBCI_Ganglion.<init>(OpenBCI_GUI.java:8440)
	at OpenBCI_GUI.systemUpdate(OpenBCI_GUI.java:774)
	at OpenBCI_GUI.draw(OpenBCI_GUI.java:418)`

After what happened last time (mistakenly using the USB dongle) I have a bad feeling that there is something obvious that I am missing.

I will test this on a 3rd mac once I get a chance.

aj-ptw, 2017-02-16:
Does your iMac support Bluetooth 4.X? AKA BLE?

matthew-groth, 2017-02-16:
Apple Bluetooth Software Version:	4.3.6f3 16238
  Bluetooth Low Energy Supported:	Yes

I'm about to update my issue on Node.JS with the results from the iMac. which are definitely interesting. Then I have to go home and won't be able to test anything on this device again until tomorrow.

matthew-groth, 2017-02-16:
Also see the issue I created concerning the same iMac on Node.js [here](https://github.com/OpenBCI/OpenBCI_NodeJS_Ganglion/issues/22).

aj-ptw, 2017-02-16:
@matthew-groth 
 `FTDIUSBSerialDriver_v2_2_18.dmg` is only for the Cyton board

> I definitely put the ganglion hub into the data folder on the imac

Where did you get the ganglion hub from? Did you build it yourself?

matthew-groth, 2017-02-17:
@aj-ptw 

After reading this I checked and it turns out I had the wrong version (3.0 instead of 4.1). I'll make an issue in the docs, which mistakenly linked the 3.0 version. I updated to 4.1, and OpenBCI GUI works on the iMac!

aj-ptw, 2017-02-17:
@matthew-groth thank you so much for pointing this out! I'm so sorry this happened to you, I hope others find this! I'll go hit up the docs

aj-ptw, 2017-02-17:
@matthew-groth where in the docs is the issue? mind opening  an issue? you have been a great help

closed 2017-02-17 by aj-ptw

matthew-groth, 2017-02-17:
Yes, just an old link. Sorry I said issue, but I actually [changed the link in a PR](https://github.com/OpenBCI/Docs/pull/70). 

Thank you so much to your team for open sourcing your hard work. You're the one really helping me!