OpenBCI/OpenBCI_GUI #132: MacOSX download link has a "damaged" file

jessicaquynh, 2017-02-17:
Running this on OSX 10.11.16. Amd I am getting an error when trying to launch the OpenBCI_GUI v2.0.0 (last updated Jan 2017) executable:

`"OpenBCI_GUI" is damaged and cannot be opened. You should move it to the Trash.`

Has anybody else gotten this issue?

jessicaquynh, 2017-02-17:
Apologies, I realize there was an open issue on the discussion forum: http://openbci.com/index.php/forum/#/discussion/664/on-mac-openbci-gui-is-damaged-and-can-t-be-opened

Which points to the classic unidentified developer permissions error (and resolved my issue). Though, is there anyway to make that more clear to users?

aj-ptw, 2017-02-17:
Yes, figure out why that dang application won't sign!

closed 2017-02-17 by aj-ptw