OpenBCI/OpenBCI_GUI #133: GUI requires running with sudo to see Ganglion bluetooth on Ubuntu Linux

prefrontalvortex, 2017-02-17:
Running Ubuntu 16.04. I have to use sudo ./OpenBCI_GUI in order to see the Ganglion on the bluetooth device list, even though I am added to the bluetooth and dialout groups. I can run and collect fine doing it this way, but I feel like I shouldn't need sudo to get this to work. Feels yucky. 

Any thoughts how to remedy this?

I'm using the built-in bluetooth on my laptop, but the result is the same with the CSR dongle.

rodrigohubner, 2017-02-23:
You need to use "sudo" because the permissions of your USB port (dongle device) is not sufficient.
Type "sudo chmod 666 /dev/ttyUSBn".
Or, permanantly: http://unix.stackexchange.com/questions/25258/ttyusb0-permission-changes-after-restart

aj-ptw, 2017-03-16:
@rodrigohubner is there something we can do from an application perspective to make this work? Thanks! Closing for now unless we figure out what we can change on our end!

closed 2017-03-16 by aj-ptw

rodrigohubner, 2017-03-17:
On Linux: I don't know anything that can be done by software.