OpenBCI/OpenBCI_GUI #135: Could not connect open bci with open vibe acquisition server

AshmitDadhich9696, 2017-02-23:
Hello

We are facing problem in interfacing Openvibe with OPenBCI Board. Openvibe with Generic oscilattor is working good and we getting signal in OPenBCI GUI from our open bci board. We also did driver settings and changes in Preferences.But unable to cope up with problem.We also followed instructions given in documentation.

Error : No response from the driver in 1000ms
here by i am providing the screen shot of the terminal please help me out
[image]

aj-ptw, 2017-03-16:
Hi are you able to use the processing GUI?

aj-ptw, 2017-03-16:
Closing because you are saying it's working in this GUI. Could you route the question over to the OpenVibe team? We don't use the program in house. Sorry this would frustrate me too!

Thanks!

closed 2017-03-16 by aj-ptw