OpenBCI/OpenBCI_GUI #138: Modify GUI Headmap Electrode Positions

CelticHarp, 2017-02-25:
It would be nice the file 'electrode_positions_default.txt' to be editable and readable by the GUI in order to change the default electrode positions that correspond to the GUI headmap.
Maybe at a later improvement we could choose from a picture the positions and the file to be updated automatically with the coordinations.
In any case a documentation of this capability would be essential.

Reported by: compmonks, jussijaatinen, celticharp

Thank you!

compmonks, 2017-03-02:
yes, it would be nice to have a text file with 10-20 coordinates adjusted for the GUI along with a way to label electrodes with their correct name (Fz, FCz, Oz, O3, ...) instead of their index (1,2,3,...). I guess most users would find it also very useful  and time-saving :)

aj-ptw, 2017-03-16:
@conorrussomanno wrote some code for this once upon a time

aj-ptw, 2017-06-02:
closed by #152

closed 2017-06-02 by aj-ptw

emolyzr, 2018-02-07:
Hi,
thanks for adding the drag and drop option. Unfortunately, this doesn't solve the issue. One can move the electrode positions, but this won't affect the calculation of the scalp distribution in any meaningful way.
[image]

In addition, it's not possible to save the configuration, so that upon reloading the GUI one has to move the nods again.

Any help would be highly appreciated.

emolyzr, 2018-02-07:
As a workaround to solve this issue: 
1.) Just go to ~\OpenBCI_GUI\data 
2.) Open the file electrode_positions_default.txt and change the settings according to your setup (You may wish to make a backup of the original file first). 
3.) When starting the GUI, the head plot widget will always read out this file. 

It would be convenient, to know the coordinates of the other nodes of Ultracortex Headset, however by playing around one should get the right estimates oneself.

yidilozdemir, 2020-11-05:
I have the latest version of the OpenBCI_GUI, and here I cannot find this location ~\OpenBCI_GUI\data. What is the current fix for the issue described?

PR #159 (2017-06-02): Development
# 2.2.0

### Bug Fixes
* Fix #151 - Incorrect number of channels on playback caused index out of bounds errors.
* Addresses #149 - Allows for proper scaling of channels with four thanks to #151 #157

### New Features
* Band power widget #153 (thanks @sunwangshu)
* Closes #138 - Able to drag and drop the electrodes on the head map  (thanks @liqwid)