OpenBCI/OpenBCI_GUI #145: GUI is unusable on high DPI screen

patrickmineault, 2017-03-17:
I use a Dell XPS 13 on Ubuntu, which has a retina-like display (3200x1800). The OpenBCI GUI doesn't scale appropriately, and the interface elements are so small as to be unusable:

[image]

 This is a common occurrence on Java and Processing-based programs, and most programs (Processing, Arduino, RStudio) have some setting to scale the interface. In the case of Arduino, it pretty much works (shown here at 200%), although Processing only partially scales the interface. Here are some related bugs from Processing:

https://github.com/processing/processing/issues/4183
https://github.com/processing/processing/issues/4895

Just adding a setting to scale the interface would work, I think.

aj-ptw, 2017-03-24:
@patrickmineault thanks for the report! This is an important fix, will dig in.

aj-ptw, 2017-03-24:
whoops did not mean to close that

aj-ptw, 2017-06-09:
This is bigger than I though to implement. Working to figure out how long it will take!

aj-ptw, 2017-06-27:
Because the upstream processing is taking forever to fix this, going to address this...

aj-ptw, 2017-06-27:
@conorrussomanno I made some progress on this, there is a problem right now with mouse hover over. Can you please boot up this version of the GUI and adjust the scale factor, you'll see how to do it really quickly, it's self explanatory. Currently need to move the hover over when buttons update their size. If you want to use the new function `guiScale(float)` or `guiScaleInt(int)` call them in the draw loops of the widgets and such, not on initialization!

aj-ptw, 2017-09-07:
@patrickmineault there is an option if you right click on the executable -> properties -> compatibility -> don't do high DPI scaling (or something similar) that should solve this problem for now!

daniellasry, 2019-04-28:
The root of the issue is that [processing uses java 8, and it's not looking like they will update it anytime soon.](https://github.com/processing/processing/wiki/Supported-Platforms#java-versions)

Java 8 is telling the OS that it is DPI aware, [but doesn't scale UI elements](https://superuser.com/questions/988379/how-do-i-run-java-apps-upscaled-on-a-high-dpi-display).

On windows, we can work around the issue by [forcing the OS to scale the app](https://github.com/OpenBCI/OpenBCI_GUI/pull/503).
(it will look blurry but readable)

On linux, there is no way to force scaling. The hack I found that works, is using this script:
https://github.com/kaueraal/run_scaled

retiutut, 2019-05-17:
See @daniellasry's comments above. Closing this issue.

wjcroft, 2019-05-19:
@retiutut and @daniellasry , the current 'footnote' in the GUI doc says only this, very tersely:

> If you have a high DPI screen and are running windows 10, you may notice that the GUI is hard to see, and nothing seems to scale correctly. Select Change high DPI Settings and on the bottom drop down box, change from Application to System (Enhanced).

And that statement has apparently confused at least one high DPI user (see post near bottom):

http://openbci.com/forum/index.php?p=/discussion/2128/can-the-gui-playback-recordings-gui-on-high-dpi-screens

Here is an illustrated article showing the setting in detail:

https://www.windowscentral.com/how-improve-app-dpi-scaling-enabling-system-enhanced-feature-windows-10

Is this something we should link to from the docs page? I have not heard back from the user on the above thread.  Not sure if this will solve his issue yet. He claims on the forum thread that he tried to follow directions, but not sure if he connected the dots to open the Properties on the GUI app file.

daniellasry, 2019-05-19:
hey @wjcroft ! That's supposed to be fixed in 4.1.2 (and still looks fixed after a quick test). I'm puzzled to why this user still had the issue with 4.1.2. I'll go ahead and ask for a screenshot.

wjcroft, 2019-05-19:
Daniel, thanks much. Apparently zfmeng was able to adjust the Properties dropdown to fix it. I asked him to repost his screenshot from before the fix. I think zfmeng's confusion was mostly over our terse instructions in the docs file. It does not mention the GUI exe explicitly, nor that this is done on the Properties dialog box. Thanks again for following up on the forum. You should get an email notification if/when he reposts his screenshot.

retiutut, 2019-05-20:
The Docs have been updated to reflect the information in the Issue. Also included pictures for Windows users.

wjcroft, 2019-05-20:
Richard (and Daniel), thanks so much!

> One nitpick, your text in the Docs page starts out: "If you have a high DPI screen and are running Windows 10, you may notice that the GUI is hard to see, and nothing seems to scale correctly. Right click on OpenBCI_GUI.exe and select Properties. The following window should pop up: ..."

However, zfmeng on the forum had the problem that he attempted to change the Properties WHILE the GUI was running. The GUI should be closed if running (and hard to see / poor scaling). THEN the Property box changed, then the GUI restarted.

Zfmeng stated clearly that he changed the property back and forth and did not see any changes. This is why.

Regards, William

retiutut, 2019-05-20:
@wjcroft Ok that's the last part. Still need to add that but I have to leave my desk right now.

retiutut, 2019-05-21:
Ok updated the docs to say close the GUI first.

wjcroft, 2019-05-21:
Hey, thanks much Richard. I was surprised when Zfmeng was at first not able to adjust the Properties, even after giving him the Windows Central article link. So being this explicit will help future users.

retiutut, 2019-05-21:
Could even be clearer if the screenshots start having annotations (easy to do on Mac). Closing this issue for now, unless more information arises.

closed 2019-05-21 by retiutut