OpenBCI/OpenBCI_GUI #146: OpenGL error at bot beginDraw()

MirzaUbaidullahBaig, 2017-03-24:
[image]

aj-ptw, 2017-03-24:
Are you able to run any processing file?

MirzaUbaidullahBaig, 2017-03-24:
I just ran some basic example codes and one that i used with arduino in processing to be sure. there is no problem in running. 
[image]

aj-ptw, 2017-03-24:
do the examples use OpenGL?

MirzaUbaidullahBaig, 2017-03-24:
i don't know if you know any example that use openGL, give me.

aj-ptw, 2017-03-24:
The processing GUI uses OpenGL, that's the only example I can provide. I'm not familiar with what it takes to run OpenGL on raspberry pi

MirzaUbaidullahBaig, 2017-03-24:
I'll google some examples. w8

MirzaUbaidullahBaig, 2017-03-24:
There are 3 graphic drivers in raspberry pi one is default non-gl driver, others are experimental GL drivers. I enabled one of those drivers.
[image]
still got the same error.
Please help me.

aj-ptw, 2017-03-24:
Maybe this application is too much for raspberry pi, not sure there is anything we can do to help you.

closed 2017-05-18 by aj-ptw