OpenBCI/OpenBCI_GUI #148: LSL connection doesn't send streaming correctly

rodrigohubner, 2017-04-17:
The follow error ocurred when I execute the receive code:
```
$ java -Djna.nosys=true -cp "jna-4.2.2.jar:." examples.ReceiveData
Resolving an EEG stream...
Note: The stream named '(invalid: bad lexical cast: source type value could not be interpreted as target)' could not be recovered automatically if its provider crashed because it does not specify a unique data source ID.
Error while receiving the stream info (Invalid argument); retrying...
edu.ucsd.sccn.LSL$LostException: The stream has been lost.
	at edu.ucsd.sccn.LSL.check_error(LSL.java:900)
	at edu.ucsd.sccn.LSL$StreamInlet.info(LSL.java:568)
	at edu.ucsd.sccn.LSL$StreamInlet.info(LSL.java:569)
	at examples.ReceiveData.main(ReceiveData.java:14)
```
I have the problem using "send" in java and "receive" in everything else language.

aj-ptw, 2017-04-17:
Post code that you are using, the parser is not able to get the string name. So there is something wrong with the way you are using stings. Or post a screen shot. Are you using " in the call?

rodrigohubner, 2017-04-17:
I connected the send data with GUI.
The code in python is the basic example on https://github.com/sccn/labstreaminglayer/blob/master/LSL/liblsl-Python/examples/ReceiveData.py
So, when I start the send data with LSL connection on GUI and run the script shows this error:
```
$ python ReceiveData.py 
looking for an EEG stream...
Note: The stream named '(invalid: bad lexical cast: source type value could not be interpreted as target)' could not be recovered automatically if its provider crashed because it does not specify a unique data source ID.
Traceback (most recent call last):
  File "ReceiveData.py", line 10, in <module>
    inlet = StreamInlet(streams[0])
  File "/home/rhubner/.anaconda2/lib/python2.7/site-packages/pylsl/pylsl.py", line 641, in __init__
    self.sample = self.sample_type()
TypeError: 'list' object is not callable
```

rodrigohubner, 2017-04-17:
@aj-ptw Is there some way to create a script in Python to send LSL stream from OpenBCI GUI?

Because honestly, the biggest problem for me is with Java. I just wanted something using OpenBCI to send LSL data showing the signal in time...

aj-ptw, 2017-04-17:
@rodrigohubner yes this example was written by @gabrielibagon https://github.com/OpenBCI/OpenBCI_NodeJS/tree/master/examples/labstreaminglayer

It's a very stable driver.

rodrigohubner, 2017-04-17:
But it would be better to see the EEG signal in time on a graph! OpenBCI GUI is great but not works the LSL send for me...

aj-ptw, 2017-04-17:
totally get what you are saying! it's a good visual!

rodrigohubner, 2017-04-17:
I didn't know how to do it using OpenBCI_NodeJS with LSL. Can you explain the steps to me by e-mail, please? (off topic to this issue). rhubner@gmail.com

aj-ptw, 2017-04-17:
Sorry! I didn't write this part so all i can do is point you to the instructions found here https://github.com/OpenBCI/OpenBCI_NodeJS/blob/master/examples/labstreaminglayer/readme.md
If the instructions are unclear, please open an issue in that repo. Thanks so much!

aj-ptw, 2017-04-17:
@rodrigohubner you should check out the development branch https://github.com/OpenBCI/OpenBCI_GUI/tree/development look like gabe added LSL support on the last commit? not sure @conorrussomanno may know better

aj-ptw, 2017-07-19:
fixed on master AFAIK, please reopen if issues persist

closed 2017-07-19 by aj-ptw