OpenBCI/OpenBCI_GUI #149: Playback for ganglion record's looks noisy

likeleto, 2017-04-21:
Started GUI in playback mode for the record from ganglion. Discovered that time series graph looked very noisy compared to live stream.
Here the comparison
[image]

aj-ptw, 2017-05-18:
Hmm this is interesting. Did you try selecting four channels instead of eight?

likeleto, 2017-05-18:
@aj-ptw  that's the thing, 4 channels was chosen, but in playback, it looks like in the left picture. So I might guess that GUI in playback mode doesn't know anything about ganglion and think that it is 8ch cynton.

aj-ptw, 2017-05-18:
what version of the GUI are you using?

likeleto, 2017-05-18:
@aj-ptw mac, 2.1.2 (from site and through processing, same behavior)
attach example of record from my ganglion
[OpenBCI-RAW-2017-04-15_01-14-57.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/1012552/OpenBCI-RAW-2017-04-15_01-14-57.txt)

aj-ptw, 2017-05-31:
@likeleto just tried running and the data looks pretty good http://g.recordit.co/CzjHkjbROb.gif

I used a patched version of the GUI that allows for four channels in playback see pr #157 which was started because of #151 

If that's EEG then it's pretty good, your channel 1 looks very noisy, try a different electrode?

likeleto, 2017-05-31:
@aj-ptw Thank you for patch, I've just tested updated version. There're is still the difference. Why does playback graph not show the same picture as the live graph?
**LIVE**
[image]

**PLAYBACK**
[image]
Thanks again for your help.

aj-ptw, 2017-06-02:
is it possible that the filters are not being set? please see if the notch and band pass are on!

likeleto, 2017-06-02:
Notch and BP were set, settings on pictures the same...

aj-ptw, 2017-06-27:
Have you reached out to @biomurph ?

daniellasry, 2019-01-05:
Closed by #159

closed 2019-01-05 by daniellasry