OpenBCI/OpenBCI_GUI #150: Issue Running GUI from source

lYlantis, 2017-04-28:
When I download the repo and try to run from source, I am getting this error message:

[image]

Even though in libraries is GifAnimation and UDP which I assumed was the other library I needed.

[image]

I tried dragging and dropping the folder into the sketch folder as mentioned in an earlier issue. I am on Windows 10.

aj-ptw, 2017-05-18:
Hi is this issue still persisting?

Are your preferences set correctly inside processing? Prefresences -> Sketchbook location -> Folder where all these libs are placed.

aj-ptw, 2017-06-02:
Hi please let us know if this issue is still outstanding! Closing for now.

Thanks!

closed 2017-06-02 by aj-ptw