OpenBCI/OpenBCI_GUI #151: Cannot replay saved sessions

nateGeorge, 2017-05-15:
There seems to be an error when opening a PLAYBACK from SD

```
User selected /home/nate/Desktop/all/standalone programs/application.linux64/SavedData/Nate Eye experiment-RAW-2017-05-15_17-41-02.txt
Timing SD file conversion...

%OpenBCI Raw EEG Data

%

%Sample Rate = 250.0 Hz

%First Column = SampleIndex

%Last Column = Timestamp 

%Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data

java.lang.NumberFormatException: For input string: "0000 1574.28"
	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
	at java.lang.Long.parseLong(Long.java:589)
	at processing.core.PApplet.unhex(PApplet.java:9857)
	at OpenBCI_GUI.convertSDFile(OpenBCI_GUI.java:5227)
	at OpenBCI_GUI$ControlPanel.update(OpenBCI_GUI.java:1727)
	at OpenBCI_GUI.systemUpdate(OpenBCI_GUI.java:791)
	at OpenBCI_GUI.draw(OpenBCI_GUI.java:418)
	at processing.core.PApplet.handleDraw(PApplet.java:2418)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:898)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
```

aj-ptw, 2017-05-18:
Hey! Sorry this sounds super frustrating!

Could answer these questions so we can help you a little better?

What version of the GUI are you using?
What operation system are you using?
What board is this? 
Can you please share the SD file?
What speed class is the SD card?

Thanks so much!

nateGeorge, 2017-05-18:
Hey thanks AJ.  

version 2.1.2 (latest I think)
OS Ubuntu 16.0.4 LTS
Cyton Board with 16-channel expansion

It's actually not an SD file, it's just a normal data log file.  I can share it if you want.  Is it not possible to replay those?

aj-ptw, 2017-05-18:
Please do share it

nateGeorge, 2017-05-27:
So the error only happens when trying to load the file as SD.  When loading as a 'Playback File', nothing happens.  Here is one file: https://www.dropbox.com/s/cen2viz5yba0d3w/Nawar_eyes_open_closed_OpenBCI-RAW-2017-05-15_18-05-29.txt?dl=0

aj-ptw, 2017-05-31:
Hey @nateGeorge just downloaded the file. Checking it out now

PR #159 (2017-06-02): Development
# 2.2.0

### Bug Fixes
* Fix #151 - Incorrect number of channels on playback caused index out of bounds errors.
* Addresses #149 - Allows for proper scaling of channels with four thanks to #151 #157

### New Features
* Band power widget #153 (thanks @sunwangshu)
* Closes #138 - Able to drag and drop the electrodes on the head map  (thanks @liqwid)
closed 2017-06-02 by aj-ptw