OpenBCI/OpenBCI_GUI #154: Cyton working with old OpenBCI_GUI, but not with recent OpenBCI_GUI (v2.1.2)

Wmanns, 2017-05-27:
After some while of not using Cyton I recently updated the dongle driver (**FTDI version 2.12.26.0**) and the OpenBCI_GUI (**v2.1.2**). I followed carefully the instruction (**Cyton Getting Started Guide**) but Cyton was not connecting. I tried the "OpenBCI Radio Configuration Utility" (http://openbci.com/community/openbci-radio-configuration-utility/). Here the COM port (#37) is listed, but after selecting it, there is no reaction whether a connection could be established or not.

Using an old GUI [needs Processing 2. No vs number in code. But heading states: "GUI for controlling the ADS1299-based OpenBCI // Created: Chip Audette, Oct 2013 - May 2014 // Modified: Conor Russomanno & Joel Murphy, August 2014 - Oct 2014"] . 
With this old GUI Cyton connected without problems and transmitted valuable data.

What do I have to do to be able to work with the new GUI?

Thanks a lot!
Will

System: Windows 7. 16 GB.
Original post in forum: [http://openbci.com/forum/index.php?p=/discussion/1136/cyton-working-with-old-openbci-gui-but-not-with-recent-openbci-gui-v2-1-2](url)

aj-ptw, 2017-05-31:
Hi sorry it's not working for you! 

Did you try using Processing 3.x to run the GUI? The new GUI requires running the newer version of processing.

Wmanns, 2017-05-31:
It's the GUI (most probably a compiled one?) from: http://openbci.com/index.php/downloads, the windows 64 bit version. I do not see a version number, the size is: 138.980.380 Byte. My system is a 64bit windows 7 12 GB PC. Thank you for your interest! Hope that helps.

aj-ptw, 2017-06-02:
Can you please try running as an Administrator? See #155

Wmanns, 2017-06-04:
Thank you, your hint helped: 
When I installed OpenBCI_GUI.exe I copied the folder "application.windows64" into "C:\Program Files\OpenBCI\". Then I made 
a link from "_C:\Program Files\OpenBCI\application.windows64\OpenBCI_GUI.exe_" to my desktop. 
Starting OpenBCI_GUI.exe from this link - even as admin - the GUI freezed. 
But moving the contents of "application.windows64" to a folder "C:\Program Files\OpenBCI\" then making a link from "_C:\Program Files\OpenBCI\OpenBCI_GUI.exe_" to the desktop 
and starting "OpenBCI_GUI.exe" as admin finally worked. 

My _**working installation path**_ (under windows 7, **_running as admin_**) is :
"_**C:\Program Files\OpenBCI\OpenBCI_GUI.exe**_"
Thank you, aj-ptw!

closed 2017-06-04 by Wmanns