OpenBCI/OpenBCI_GUI #155: "Failure: Communications timeout - Device failed to poll Host" Autoscan not working (in Windows?)

MCube78, 2017-05-29:
**Short story:** After receiving the latest OBCI Cyton&Daisy hardware I followed the installation's instruction, but after manually changing the channel (as per the tutorial), I cannot seem to get the board and dongle back in communication. Currently I seem to be stuck at "Communications timeout - Device failed to poll Host". (More detail below on the system and what I've done)

**Detail:**
_System_: ASUS Win 10 laptop (same issue on Dell XPS Win 10 -- which is where the issue started).
_OBI hardware_: latest Cyton&Daisy

_Where the issue started_: I received the OBCI hardware, followed the instructions, installed the latest drivers, downloaded the OBCI GUI, could see my COM port, and all went well up to the point where the tutorial demonstrates how to manually change the channel and then fix it by using autoscan. The second before doing that when I pressed "status" (as shown in the tutorial) and it return something to the effect that the system was up and the dongle and device were talking on a certain channel (can't recall if 15 or 20, either way, system was up and the two seemed to communicate just fine). After manually changing the channel, when I pressed Autoscan as suggested in the tutorial, that didn't fix the issue, and I couldn't get the two to talk again.

_What I've tried so far_: (1) I tried to manually select the channel, tried with the override dongle too, tried autoscan a few more times, but no luck. I found a post on a similar error (http://openbci.com/forum/index.php?p=/discussion/comment/6194#Comment_6194) so tried following the instructions. (2) I've installed the latest version of Processing (3.3.3), ran the OBCI GUI from there, but no luck with the autoscan from that version of the OBCI GUI (other than I could start seeing the errors in the debugger). (3) I downloaded the Radio Config utility (as suggested in the post above), tried the autoscan etc from there, but still no luck. Whatever I do I cannot seem to get around the "Failure: Communications timeout - Device failed to poll Host$$$" error. (4) I tried accessing my COM from a shell (I used both Putty and the windows powershell, but while my COM is listed, and indeed I can see it from the Windows device manager, I get the same error as above).

aj-ptw, 2017-05-29:
This sounds super frustrating! 

Thank you so much for the detailed report. I'm going to make sure you get this working.

Can you reprogram and RFDuino on the Cyton? Over ride the channel as stated in the radio programming tutorial? Set both to the same channel and reprogram the Dongle too. Remember the radio channel is stored go FLASH and is retained even after a power cycle. 

Let's get you data!

aalash, 2017-05-29:
Greetings AJ and thanks for the followup. 

Since I’m not a computer guy, can you send step by step instructions?

Thanks,
aa

aj-ptw, 2017-05-29:
http://docs.openbci.com/Hardware/06-Cyton_Radios_Programming_Tutorial

aj-ptw, 2017-05-29:
You should be able to recover this though...

aj-ptw, 2017-05-29:
Without having to reprogram the radios. Can you make sure your board is powered and hour auto scan again? Keep the radios within 1 metres for this part

aalash, 2017-05-29:
Thanks for your help Aj.

I’ve decided to give up on my PC.

1. Now my MAC: although the system is up when I press START SYSTEM, the gui freezes.  Can you help?

2. Re connecting 16 colored wires to daisy module, this article http://docs.openbci.com/Tutorials/01-Cyton_Getting%20Started_Guide#cyton-getting-started-guide-v-connect-yourself-to-openbci-2-connect-your-electrodes-to-openbci <http://docs.openbci.com/Tutorials/01-Cyton_Getting%20Started_Guide#cyton-getting-started-guide-v-connect-yourself-to-openbci-2-connect-your-electrodes-to-openbci> only discusses the 8 channel board —even the wires used have different colors than of the Ultra-Cortex Mark IV.  

I know you guys are super busy but I have nowhere else to go for help --Thank yo,
aa

aalash, 2017-05-29:
I got the PC to work with overriding the channel; auto-scan no go.  I’m fine with that. 

I’m experiencing the same symptom as on the MAC where the software freezes as soon as I click Start System. Pls see pic:

aalash, 2017-05-29:
restarted commuter and all worked fine. thanks. 

Only thing left is to connect the 16 colored cords to the daisy module -no clear instructions.

MCube78, 2017-05-29:
Hi AJ,

 thank you for the answer. Yeah, frustrating but I know it's the way things go, particularly when you are not proficient with something -- also because I am only waiting on getting this setup to start data collection for a new experiment.

Anyways, I followed the tutorial - to a T - but no luck..
Everything goes just fine up to when I try to upload/flash the memory. When I tried flashing the memory, it verified just fine, but then when I pressed upload it gave me an error: 
"expecting 14, got 46
fail.......fail.......fail......."
and then just stalls.

 Is there something wrong with how I flashed it (here is the setup function after I inserted the flashing command):
void setup() {
  // Declare the radio mode and channel number. Note this channel is only
  //  set the first time the board powers up OR after a flash of the non-
  //  volatile memory space with a call to `flashNonVolatileMemory`.
  // MAKE SURE THIS CHANNEL NUMBER MATCHES THE HOST!
  radio.flashNonVolatileMemory();
  radio.begin(OPENBCI_MODE_DEVICE,20);
}

There is, however, one thing about the tutorial that I could not interpret, so it might be that. At some point it says:
"If you want to modify the firmware that the OpenBCI Dongle came with, or roll your own, make sure that you are setting the RFduino up as a DEVICE, and that channel is selected correctly."
Now, unfortunately my response to this slightly cryptic sentence is sure, but "where?" and "how?" 0_o

heeeeeelp.. ;-)

 Cheers

 Martin

MCube78, 2017-05-30:
AJ, so I've kept going with the tutorial, and managed to successfully flash and reupload the firmware to the dongle, and after that a few things change (but the crucial one didn't...):
1. If I now try to run the RadioDevice32bit (with or without the flash command) uploading to cyton (i.e., with GPIO_6 selected instead of reset), the error is slightly different (but equivalent I'd say..):
"get
fail.......fail.......fail......."
2. Now if I try running the OBCI GUI (I do it from processing), and click status it says "Error reading from the serial port. Try a different port?"

aj-ptw, 2017-05-30:
You need to use version 2.0.0 NOT 1.0.0 tutorials!

MCube78, 2017-05-30:
Yep, these: [http://docs.openbci.com/Hardware/06-Cyton_Radios_Programming_Tutorial#cyton-radios-programming-tutorial-uploading-device-firmware-to-cyton-board-upload-device-radio-firmware-version-2xx-fall-2016]

aj-ptw, 2017-05-31:
My apologies, I got worried the docs were misleading. Can you please email info @ push the world . us (no spaces) and we can set up a time to chat.

aj-ptw, 2017-06-02:
Fun talking with you Martin! I'm so pleased we were able to resolve all of your issues. The big problem with not being able to upload was having the radio config open with the OpenBCI_GUI, it had a lock on the serial port and prevented Arduino from being able to upload code. The problem with the GUI freezing on your computer was because the first time you ran the GUI, it tried to make a folder called 'SavedData' which resulted in the refreezing of the GUI. You solved that by running the OpenBCI_GUI as an administrator. I hope you the best of luck and please let us know if you ever have any problems in the future!

closed 2017-06-02 by aj-ptw

MCube78, 2017-06-03:
Thank you very much AJ!

One small note: In windows you actually have to run the GUI as admin every time because it will try to create a Data txt file, not just the first time you launch it (when it tries to create the SavedData folder).