OpenBCI/OpenBCI_GUI #163: FTDI VCP Version Compatibility with OpenBCI_GUI on Mac OS

DaveHChen, 2017-06-11:
As requested by http://openbci.com/forum/index.php?p=/profile/570/wjcroft
I'm logging this issue for AJ @PushTheWorld and others to see.

Today, I was able to get live data streaming from Ganglion to Mac running Processing, after following FTDI's doc to remove the existing VCP and to install version 2.2.18.
By the way, version 2.4.2 is not compatible.

http://www.ftdichip.com/Drivers/VCP.htm

aj-ptw, 2017-06-12:
What was the problem?

DaveHChen, 2017-06-12:
Before installing version 2.2.18, I was unable to stream live data from Ganglion to Mac.

aj-ptw, 2017-06-12:
The ganglion does not use VCP though or an FTDI driver.

DaveHChen, 2017-06-12:
Then, I've no ideas how the VCP downgrade fixed my data streaming problem.
Anyway, I'm running MacOS 10.12.5 on a MacBook Pro.
And, I'm running OpenBCI_GUI through Processing with many libraries, modes, and tools installed.

aj-ptw, 2017-06-12:
Hmm did you drag and drop the `GanglionHub`? Going to close this for the meantime

closed 2017-06-12 by aj-ptw