OpenBCI/OpenBCI_GUI #166: Run it, But it show: please fix the size() line to continue.

neotim, 2017-06-14:
[image]
[image]

Run it, But it show: please fix the size() line to continue.

aj-ptw, 2017-06-14:
@neotim 

Thanks for using and supporting the OpenBCI GUI!

_This issue seems like a missing dependency_

Did you configure the libraries correctly? (Show a screen shot of your preferences page in processing, I want to see your `Sketchbook folder:`. Then take a screen shot of that finder window where the `Sketchbook folder` is pointing to)

What version of Mac OS are you running?
What version of the GUI are you trying to run? (when did you download it?)

Thanks!

neotim, 2017-06-14:
Thank you for your reply, these are screen shots:
[image]
[image]
[image]

// GUI for controlling the ADS1299-based OpenBCI
// Created: Chip Audette, Oct 2013 - May 2014
// Modified: Conor Russomanno & Joel Murphy, August 2014 - Dec 2014

aj-ptw, 2017-06-14:
Can you please show me a screen shot of where `/Users/devupload/Documents/Processing`

Mine looks like this:
[image]

neotim, 2017-06-14:
ok, this is the screen shot of /Users/devupload/Documents/Processing:
[image]

aj-ptw, 2017-06-14:
Can you please download processing 3.3 **NOT** 3.3.4 and let me know if that helps.

neotim, 2017-06-14:
now is processing 3.3, same error:

[image]

aj-ptw, 2017-06-14:
So it looks like you have the new version of the GUI inside the old version the GUI. You should clone the gui into a new directory. The size error is coming from the old gui code...

neotim, 2017-06-14:
yes, this old gui is from https://github.com/viguix/OpenBCI_GUI_Helicopter_Throttle
need to control copter.

aj-ptw, 2017-06-14:
I think that was done using processing 2.x not 3.

Closing this issue

closed 2017-06-14 by aj-ptw

neotim, 2017-06-15:
now is processing 2.2.1: 
[image]

aj-ptw, 2017-06-15:
Glad to see you opened on @biomurph repo. He's good and will help. Good luck!

ronnyandre, 2019-01-24:
Same error with Processing 3.5.2

Redha-Ferradji, 2019-03-24:
you just need to replace size () with surface.setSize () and for more information
https://github.com/processing/processing/wiki/Changes-in-3.0