OpenBCI/OpenBCI_GUI #172: Issue with Open BCI GUI

AnkitaKhadpekar, 2017-07-06:
Hello,
I want to do one project in which I need to acquire bio signals (EEG and EMG). I am planning to use open BCI V3-32 cyton board. But I am facing some problems with open BCI GUI. So I need some help to resolve following issues:

When I launch GUI, in radio configuration in status I am getting error as "Error connecting to serial port"
In spite of this error, when I connect dongle and cyton board some random signal is getting transmitted. TXD (red) glows but RXD is not glowing.
When I connect dongle and then cyton board as mentioned, I am getting some random signal or noise on all 8 channels in GUI. The signal remains the same with or without connecting electrodes on these channels . There is no variation even when I touch these pins on the board
Also I have difficulty in understanding hardware settings in GUI
This I did by installing GUI as a standalone but I have not understood how to do it with processing IDE
Your help will be greatly appreciated
Thank you

aj-ptw, 2017-08-12:
Are you still having these problems?

AnkitaKhadpekar, 2017-08-12:
Hi

Now I am able to test the accelerometer and its working fine, detecting
movements in all directions. I was wondering about gyrometer to detect
orientation. But I think its not present on Open BCI board.

For bluetooth, we are planning to use ESP32 chip.

Thanks a lot,
Ankita Khadpekar

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

aj-ptw, 2017-08-12:
Okay I'm going to close this. You should run v1.0.0 of the OpenBCI_Wifi on the ESP8266

closed 2017-08-12 by aj-ptw