OpenBCI/OpenBCI_GUI #176: Wrong time stamp in linux

Jianbin-IBM, 2017-07-09:
Hi, months ago, I've raised an issue about wrong time stamp in windows which is here: https://github.com/OpenBCI/OpenBCI_GUI/issues/129
After I modify the latency and buffer, it is much better. However, when I try to collect from linux(ubunu 16.04), I saw similar problems: the time interval (last column )in the raw data seems not increase  evenly (~0.004ms). You can find it is ended 280,280,280 3 times. Any solutions? 
I used cyton board and its usb dongle.

Thanks very much!

Here is some piece of raw data.
%OpenBCI Raw EEG Data
%
%Sample Rate = 250.0 Hz
%First Column = SampleIndex
%Last Column = Timestamp 
%Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
22, -187500.02, -187500.02, -187500.02, -187500.02, 178187.92, 178172.55, -187500.02, -187500.02, 0.000, 0.000, 0.000, 08:53:17.055
23, -187500.02, -187500.02, -187500.02, -187500.02, 178161.22, 178145.92, -187500.02, -187500.02, 0.000, 0.000, 0.000, 08:53:17.070
24, -187500.02, -187500.02, -187500.02, -187500.02, 178153.88, 178138.69, -187500.02, -187500.02, 0.000, 0.000, 0.000, 08:53:17.071
25, -187500.02, -187500.02, -187500.02, -187500.02, 178168.39, 178153.09, -187500.02, -187500.02, 0.000, 0.000, 0.000, 08:53:17.072
26, -187500.02, -187500.02, -187500.02, -187500.02, 178174.53, 178159.22, -187500.02, -187500.02, 0.000, 0.000, 0.000, 08:53:17.072
27, -187500.02, -187500.02, -187500.02, -187500.02, 178155.20, 178139.80, -187500.02, -187500.02, 0.060, 0.028, 1.000, 08:53:17.077

75, -187500.02, -187500.02, -187500.02, -187500.02, 177865.61, 177850.09, -187500.02, -187500.02, 0.000, 0.000, 0.000, 08:53:17.269
76, -187500.02, -187500.02, -187500.02, -187500.02, 177869.05, 177853.56, -187500.02, -187500.02, 0.000, 0.000, 0.000, 08:53:17.280
77, -187500.02, -187500.02, -187500.02, -187500.02, 177850.67, 177835.13, -187500.02, -187500.02, 0.000, 0.000, 0.000, 08:53:17.280
78, -187500.02, -187500.02, -187500.02, -187500.02, 177825.33, 177809.69, -187500.02, -187500.02, 0.054, 0.030, 1.002, 08:53:17.280
79, -187500.02, -187500.02, -187500.02, -187500.02, 177818.42, 177802.89, -187500.02, -187500.02, 0.000, 0.000, 0.000, 08:53:17.283
80, -187500.02, -187500.02, -187500.02, -187500.02, 177836.27, 177820.77, -187500.02, -187500.02, 0.000, 0.000, 0.000, 08:53:17.289

retiutut, 2019-05-14:
The sample numbers aren't jumping, so this is just the same time stamp for 3 samples. I can't reproduce this on Ubuntu. Sometimes, I see two samples with the same timestamp. Also, there are two timestamp options now from the GUI.

I think this issue has been resolved with the additional timestamp generated for OpenBCI playback files.

retiutut, 2019-05-14:
Maybe there is some way to make the timestamps more accurate.

wjcroft, 2019-06-10:
You can be assured that if the sample number is advancing sequentially, that each data collection is happening at the 250 Hz rate. The "time stamp" variations are just due to CPU processor response latencies on the laptop machine. The time stamp is not "wrong".

closed 2019-06-10 by retiutut