OpenBCI/OpenBCI_GUI #178: Ganglion not listing BLE devices

gilesforrest, 2017-07-16:
My Ganglion is visible via BLE on other software but doesn't display in the Open BCI GUI.

E.g. using node peripheral explorer:

peripheral with ID ea0b6ab6750c found
  Local Name        = Ganglion-40ed
  Service Data      =
  Service UUIDs     =

services and characteristics:
1800 (Generic Access)
  2a00 (Device Name)
    properties  read, write
    value       53696d626c6565 | 'Simblee'
  2a01 (Appearance)
    properties  read
    value       0000 | '  '
  2a04 (Peripheral Preferred Connection Parameters)
    properties  read
    value       1000180000006400 | '    d '
1801 (Generic Attribute)
  2a05 (Service Changed)
    properties  indicate
180a (Device Information)
  2a29 (Manufacturer Name String)
    properties  read
    value       6f70656e6263692e636f6d | 'openbci.com'
  2a24 (Model Number String)
    properties  read
    value       47616e676c696f6e | 'Ganglion'
  2a27 (Hardware Revision String)
    properties  read
    value       312e302e30 | '1.0.0'
  2a26 (Firmware Revision String)
    properties  read
    value       312e312e30 | '1.1.0'
  2a28 (Software Revision String)
    properties  read
    value       302e312e30 | '0.1.0'
fe84
  2d30c082f39f4ce6923f3484ea480596
    properties  read, notify
    value        | ''
  2d30c083f39f4ce6923f3484ea480596
    properties  writeWithoutResponse, write
  2d30c084f39f4ce6923f3484ea480596
    properties  writeWithoutResponse, write
000015301212efde1523785feabcd123
  000015321212efde1523785feabcd123
    properties  writeWithoutResponse
  000015311212efde1523785feabcd123
    properties  write, notify
  000015341212efde1523785feabcd123
    properties  read
    value       0100 | ' '

aj-ptw, 2017-07-16:
Are you on Mac? Win? Linux? 

Using ble dongle?

aj-ptw, 2017-07-16:
What version of the GUI are you using?

gilesforrest, 2017-07-16:
Hi,
Using Windows 10 with a BLE dongle (installed CSR Harmony then driver replaced using Zadig)
Version of OpenBCI GUI is the newest just downloaded 2.2.

Giles

gilesforrest, 2017-07-17:
Yes it is.

gilesforrest, 2017-07-21:
Hi...any thoughts on how to help?

gilesforrest, 2017-07-21:
This is resolved - I didn't have the openBCI programme within my program files folder. Also I set it to Run as administrator.

closed 2017-07-21 by gilesforrest

aj-ptw, 2017-07-21:
So it always runs as administrator automatically? I did not know you could do that.

gilesforrest, 2017-07-21:
Yes, right click on the executable -> properties -> compatibility -> tick "Run this program as an administrator"

aj-ptw, 2017-07-21:
Cool. Thanks to you I added [this to the docs](http://docs.openbci.com/OpenBCI%20Software/01-OpenBCI_GUI#the-openbci-gui-using-the-openbci-gui-as-a-standalone-application-install-the-application):

### Install the Application

Next, you must save the application somewhere, the same way you would any other application or executable on your machine. 

#### On Mac

Place the OpenBCI_GUI in your /Applications folder. If the GUI fails to boot the first time, reboot it, a second popup will come up that asks about running the GanglionHub for the first time, please press "OK".

#### On Windows

Please place the GUI in your Program Files directory such that the structure looks like:
"\Program Files\OpenBCI_GUI\OpenBCI_GUI.exe"

Lastly, right click on the executable -> properties -> compatibility -> tick "Run this program as an administrator".