OpenBCI/OpenBCI_GUI #183: BUG: EMG over UDP produces unusable data

aj-ptw, 2017-08-07:
## Goal 
* Trying to use the UDP networking and cannot parse the packets when using EMG widget on and selecting EMG as the source

## Problem

* See the packet, but nothing that looks like a channel number or EMG data. 

## Debugging

* Able to `println` the data in terminal
* Used wireshark to verify packets were created

## Version

* Using cyton and master GUI

aj-ptw, 2017-08-07:
@conorrussomanno this bug was reported by @biomurph while in Poland

biomurph, 2017-08-10:
More info on this issue:
In Poland, we used wireshark https://www.wireshark.org/ to debug the UDP. I don't have an example of the data we got then. 
Today, I used a UDP listener written in Processing. Uses hypermedia library.
Super simple script. Here's the debug when I put only 

`void receive( byte[] data, String ip, int port ) {	// <-- extended handler
  
    print(data);
    println('\t' + ip + '\t' + port);

}`

I get the following in the terminal:

udp port 12345 listening = true
[B@45d00b2f	127.0.0.1	59607
[B@d29df1	127.0.0.1	59607
[B@cf43f2c	127.0.0.1	59607
[B@3118b8bf	127.0.0.1	59607
[B@74d444c0	127.0.0.1	59607
[B@60fd62be	127.0.0.1	59607
[B@4ddd3cd2	127.0.0.1	59607

And so on. Interesting that it has a different port number... Not sure if that's kosher or not.

When I just do 
`void receive( byte[] data, String ip, int port ) {	// <-- extended handler

    println(data);

}`

I get:

udp port 12345 listening = true
[0] 63
[1] -128
[2] 0
[3] 0
[4] 0
[5] 0
[6] 0
[7] 0
[8] 0
[9] 0
[10] 0
[11] 0
[12] 0
[13] 0
[14] 0
[15] 0

And so on. The number in the brackets increments up to [503], then rolls over to [0]. It looks like I do get non-zero numbers for the [0] to [7], but then zeros all the way to [503]. Interestingly, if I use print instead of println in the second example, I get:

udp port 12345 listening = true
[B@6a81dd67[B@2f835e53[B@2679cba[B@1ee6958a[B@7ae809d4[B@4faca11c

This is running the GUI in synthetic mode.

aj-ptw, 2017-08-12:
we must create a way to verify the networking widget is working.

aj-ptw, 2017-10-27:
This was fixed in [v3.1.0](https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.1.0) by sending JSON packets over UDP

closed 2017-10-27 by aj-ptw