OpenBCI/OpenBCI_GUI #192: Focus Widget colors

biomurph, 2017-08-11:
The colors in the focus widget make the data and bars very hard to read

aj-ptw, 2017-08-12:
@sunwangshu Could you please select a different color scheme for the focus widget background color and/or text font color? Joel and I were in love with the new widget but were had a tough time reading the text labels and such. Awesome work!

aj-ptw, 2017-08-12:
Also the popup help window you made is amazing!!!!

sunwangshu, 2017-11-02:
Hi I just found it here recently...
Got a bunch which do you guys like?
[image]
[image]
[image]

sunwangshu, 2017-11-16:
Hello I just pushed the update please check it #289

aj-ptw, 2017-11-17:
@sunwangshu awesome!! Thanks so much! Will include in the next release!

aj-ptw, 2017-12-13:
This was patched with [v3.2.0 release](https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.2.0)! Thanks so much @sunwangshu

closed 2017-12-13 by aj-ptw