OpenBCI/OpenBCI_GUI #196: No Streaming Data from Ganglion

biomurph, 2017-08-11:
Flashed a Ganglion with the DefaultGanglion firmware from the development branch of the Ganglion_Library repo. I checked with an older version of the GUI, and same issue.
Also checked with a Ganglion flashed with the current master, and it does work with the older GUI.
Here's the breakdown:

DefaultGanglion from Master Branch of Ganglion Library
Works on older GUI. Streaming and Accel
No streaming on this development branch, accel works

DefaultGanglion from Development Branch of Ganglion Library
Works on older GUI. Streaming and Accel
No streaming on this development branch, accel works

aj-ptw, 2017-08-11:
Are you saying it didn't work for you?

biomurph, 2017-08-11:
yup. looks like the problem is with the development GUI or hub. The Ganglion firmware that includes the wifi library works on the older GUI that I have, but not the newer one...

PR #199 (2017-08-24): Development
# 3.0.0

v3.0.0 set out to move **all** of the data collection to the electron hub. This means moving serial port parsing as well.

### New Features

* Able to use wifi shield with GUI. Streams in at 1000Hz for Cyton and 1600Hz for Ganglion.

### Breaking Changes

* Dependent on electron hub for all data streaming activity.

## Beta 2

Required a lot of work on the hub. But none the less, this seems to be working decently.

### Bug Fixes

* Closes #196 #195 #194 #193 #190 #188 #187 #186 #189

## Beta 1

The first beta to be released. There are some [minor issues](https://github.com [... cut]
PR #250 (2017-09-06): Development
# 3.0.0

v3.0.0 set out to move **all** of the data collection to the electron hub. This means moving serial port parsing as well.

### New Features

* Able to use wifi shield with GUI. Streams in at 1000Hz for Cyton and 1600Hz for Ganglion.

### Breaking Changes

* Dependent on electron hub for all data streaming activity.

## Release Candidate 5

Uses OpenBCIHub v1.3.0

### Bug Fixes

* Closes: #191

## Release Candidate 4

Uses OpenBCIHub v1.2.0

### Bug Fixes

* Closes: #208 - ganglion not using correct scale factor when on wifi high resolution mode
* Fixes bug where gui starte [... cut]
closed 2017-08-12 by aj-ptw