OpenBCI/OpenBCI_GUI #198: Inconsistency between seconds recording in OpenBCI_GUI and seconds of saved data in .txt file

armengol, 2017-08-19:
Hello,

Maybe I am missing something, but I am streaming data from my Ganglion using OpenBCI_GUI for 20 seconds (timing it with an external chronometer); and when I get the raw eeg data saved in the .txt file and I divide the number of samples by the sampling frequency (200Hz), I obtain around 13 seconds instead of 20 (much less).
That is, in Matlab: length(raw_eeg_data) / 200

It sounds like I am missing something but can't figure out what... any clues of what it is?

Thank you.

aj-ptw, 2017-08-19:
What os are you running? 
What version of GUI?
Can you post a seconds worth of data.

armengol, 2017-08-20:
I am running macOS Sierra 10.12.5 and OpenBCI_GUI v2.2.0 (not using the last GUI version because it didn't find my Ganglion BLE).
Find attached the .txt file saved after streaming data for 20 seconds.

[OpenBCI-RAW-2017-08-18_17-22-04.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/1237404/OpenBCI-RAW-2017-08-18_17-22-04.txt)

aj-ptw, 2017-08-21:
Can you please download again? I fixed the Mac build

aj-ptw, 2017-08-21:
I can see that you're dropping packets. Your sample numbers jump, thanks for putting the file helped a lot.

aj-ptw, 2017-08-21:
you should also try on another mac if possible and verify it's your computer that is dropping packets

armengol, 2017-08-21:
Hi, I just tried v2.2.1 on my mac and on another mac. Sample numbers still jump on both, although on the other mac much less frequently.

aj-ptw, 2017-08-21:
I wonder what version of software you're running. @biomurph may have some suggestions. How far away is your computer?

aj-ptw, 2017-08-21:
I have a similar issue on my Mac book air from 2016 but no problem on my Mac book pro. We are continually working to improve this open source driver and would love as much system information as possible. I can run a virtual Linux through virtual box and stream with no packet loss as an immediate solution you could try. Ubuntu 16.04 LTS

aj-ptw, 2017-08-21:
I had to use the ble dongle that comes with the ganglion but none the less, its a partial solution.

armengol, 2017-08-21:
I have a Mac Book pro (Mid 2014) running Sierra 10.12.5
I tried running both GUI v2.2.0 and v2.2.1. and packets drop in both.
Computer is right next to the Ganglion board (about 30 cm).

biomurph, 2017-08-21:
I don't have any suggestions right off the top. I do see the dropped packets as well in the txt file. It looks like after every 4 samples, there are two dropped packets. Do you see that to @aj-ptw ?
If there's a pattern it might help find the solution. We compress two samples into one packet, so that would mean that every third packet is being dropped!
@armengol what are you using to power your Ganglion? Can you say what version Ganglion Firmware you're using? When did you purchase your Ganglion?

armengol, 2017-08-21:
I am powering it with the 6V AA battery pack.
How can I know the Firmware version?
I purchased the board on the 27th July.

biomurph, 2017-08-21:
OK, if you purchased this past July then your frimware is up to date.
Check your mac system info and post the Hardware Features and Settings data.

armengol, 2017-08-21:
Apple Bluetooth Software Version:	5.0.4f18
  Hardware, Features, and Settings:
  Name:	**
  Address:	**
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	Yes
  Instant Hot Spot Supported:	Yes
  Manufacturer:	Broadcom
  Transport:	USB
  Chipset:	20702B0
  Firmware Version:	v137 c9260
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x05AC
  Product ID:	0x8289
  HCI Version:	4.0 (0x6)
  HCI Revision:	0x242C
  LMP Version:	4.0 (0x6)
  LMP Subversion:	0x4189
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Portable
  Composite Class Of Device:	0x38010C
  Device Class (Major):	0x01
  Device Class (Minor):	0x03
  Service Class:	0x1C0
  Auto Seek Keyboard:	On

biomurph, 2017-08-21:
Yeah, although I'm not sure what it all means. This is the dump from my macbookpro (purchased earlier this year. their latest model with the new USB ports). I also have a macbook from about the same vintage as yours. It works well with the Ganglion. Perhaps you could do a software update? Is it possible to reset your bluetooth hardware?

Apple Bluetooth Software Version:	5.0.5f1
  Hardware, Features, and Settings:
  Name:	*
  Address:	*
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	Yes
  Instant Hot Spot Supported:	Yes
  Manufacturer:	Broadcom
  Transport:	UART
  Chipset:	4350C1
  Firmware Version:	v107 c5519
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x05AC
  Product ID:	0x828E
  HCI Version:	4.2 (0x8)
  HCI Revision:	0x158F
  LMP Version:	4.2 (0x8)
  LMP Subversion:	0x616B
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Portable
  Composite Class Of Device:	0x38010C
  Device Class (Major):	0x01
  Device Class (Minor):	0x03
  Service Class:	0x1C0
  Auto Seek Keyboard:	On

armengol, 2017-08-21:
Couldn't find a Bluetooth Software update for Sierra... But I don't think this is the reason, since the other Mac I tried (iMac running Sierra) has the same Bluetooth Software Version as yours (5.0.5f1) and still jumps samples (not as frequently as with my MacBook though).

aj-ptw, 2017-08-21:
@armengol what is the `chipset` of your friends iMac? I see that you and i have `20702B0` but @biomurph has `4350C1`

My mac book air with problems:
  Apple Bluetooth Software Version:	5.0.5f1
  Hardware, Features, and Settings:
  Name:	*
  Address:	*
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	Yes
  Instant Hot Spot Supported:	Yes
  Manufacturer:	Broadcom
  Transport:	USB
  Chipset:	20702B0
  Firmware Version:	v137 c9263
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x05AC
  Product ID:	0x828F
  HCI Version:	4.0 (0x6)
  HCI Revision:	0x242F
  LMP Version:	4.0 (0x6)
  LMP Subversion:	0x4189
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Portable
  Composite Class Of Device:	0x38010C
  Device Class (Major):	0x01
  Device Class (Minor):	0x03
  Service Class:	0x1C0
  Auto Seek Keyboard:	On

armengol, 2017-08-21:
iMac's `chipset` is `20703A1`

armengol, 2017-08-22:
Any news on how to fix this?
Thanks.

aj-ptw, 2017-08-22:
I still have no fix. It's an issue on some macs that i still experience too. If you must use this mac, they i would recommend trying to run the application through Ubuntu 16.04 with VirtualBox with CSR 4.0 BLE dongle. Lastly, a new wifi shield is coming out very very soon and this will also solve your problem. I will be updating the electron hub and the under lying libraries this week so I can give you an updated version but if the issue still persists on my laptop then I'm pretty sure it will persist on yours too.

armengol, 2017-08-22:
Okay. Thank you. Let me know if you can find a fix because I'd prefer to run everything on macOS for data processing reasons...

aj-ptw, 2017-08-23:
I'm gonna try this answer about bluetooth on Mac and we can log and see maybe where the problem is! 

https://discussions.apple.com/thread/7787580?start=0&tstart=0

aj-ptw, 2017-08-23:
I wonder if the problem comes from another application using the Bluetooth at the same time... Would explain why it's so sporadic and impossible to duplicate on computers

armengol, 2017-08-23:
I don't think so because I did a bluetooth reset and unpaired all devices from my macBook...

aj-ptw, 2017-08-23:
I too did the Bluetooth reset but didn't unpair my devices. I'll try the logging later this week.

aj-ptw, 2017-08-26:
@armengol do you have any wireless devices hooked into your computer?

armengol, 2017-08-26:
I only have a usb nano receiver for wireless mouse

armengol, 2017-08-28:
While the bug is not fixed, I'll use Ubuntu in Virtualbox as you suggested. I bought the USB dongle but did not succeed in connecting Ganglion to OpenBCI_GUI (doesn't appear in BLE device list). Is there a thread/discussion in the forum talking about this issue? Couldn't find any...

armengol, 2017-08-28:
I just saw in the installation tutorial that there is this Ganglion hub that should be installed to make the connection between the GUI and the computer’s Bluetooth hardware. I have not installed the hub in my mac but still I can connect to the Ganglion board. @aj-ptw do you have it installed in your mac?

aj-ptw, 2017-08-28:
That application comes in the binary you downloaded. The CSR dongle does not work with mac os, but if you allow a USB port in virtual box, you can use the dongle plugged into your mac. Sorry this is a night mare!

armengol, 2017-08-29:
But then I do not understand why there is the section "Install Ganglion Hub on mac" ...

aj-ptw, 2017-08-29:
because the ganglion hub is not in the OpenBCI_GUI repository. When OpenBCI_GUI repository is downloaded from GitHub, it does not come with this hub. You are down loading source code.

When you download the gui from the downloads page, you are downloading binary executable which is not raw source code, so we, for convince, put the hub inside the app so mass majority of users only have to download one app.

armengol, 2017-08-29:
Oh I see. Thanks. So I guess in Ubuntu (using virtual machine) I do not need the ganglion hub if I downloaded the gui from the downloads page either. I am asking because I keep getting the error: "Code 412Error: Please turn blue tooth on" when clicking the the Refresh List button in OpenBCI_GUI running in Ubuntu...  I have bluetooth on, CSR dongle plugged and USB allowed in virtual machine... That's why I thought it might be related to the ganglion hub thing... but if not, I don't know what else to try...

aj-ptw, 2017-08-29:
Are you running the binary application?

Thanks

AJ

armengol, 2017-08-29:
I downloaded the 64-bit Linux application from the Downloads page. I unzipped the folder and simply execute the shell script on Terminal with command:  "sudo ./OpenBCI_GUI"

Thanks for your help... It's being a nightmare to get all this working...

armengol, 2017-08-29:
Solved the problem. It was an issue with virtual machine (VMWare Fusion) about sharing usb-enabled bluetooth devices between host and guest. Finally I can run the gui in Linux.

I'll work with it for now but I'd like to switch back to mac as soon as the dropping packets problem is solved, so please let me know when it is fixed. Thanks!

________________________________
From: Àlex Armengol Urpí <alex_a_u@msn.com>
Sent: Tuesday, August 29, 2017 10:30 AM
To: OpenBCI/OpenBCI_GUI
Subject: Re: [OpenBCI/OpenBCI_GUI] Inconsistency between seconds recording in OpenBCI_GUI and seconds of saved data in .txt file (#198)

I downloaded the 64-bit Linux application from the Downloads page. I unzipped the folder and simply execute the shell script on Terminal with command:  "sudo ./OpenBCI_GUI"

Thanks for your help... It's being a nightmare to get all this working...

aj-ptw, 2017-08-29:
I will!

aj-ptw, 2017-09-07:
Hi @armengol

The wifi shield will allow you to use Mac again while this problem persists 

https://shop.openbci.com/collections/frontpage/products/wifi-shield?variant=44534009550

armengol, 2017-09-07:
Okay thank you for your note. However, I'll kindly say that I don't think I should pay for an extra accessory to make it function when in the website it is claimed that the board can work with just a mac. I already paid for the CSR dongle to use it with Linux which is fine but an extra $150 is different...

https://shop.openbci.com/collections/frontpage/products/wifi-shield?variant=44534009550

aj-ptw, 2017-09-07:
@armengol 

I totally understand. I am almost done with a new Ganglion build that will hopefully remedy this situation. I will post back here once it's released. Please know this happens only to a very small amount of the macs out there! None the less, we may need to try and write some custom code for the macOS that uses a different driver, i'm spit balling here, but if we could write a native mac application that passed off the data to the GUI or hub, i think we will have success. I don't have any idea how to do that yet, but if the new ganglion hub does not solve this problem, I'm not sure what else can solve this edge case. My mac book pro works fine but my mac air does not, i think about this issue all the time and would love any support from the community in fixing this!

armengol, 2017-09-07:
Great, thanks. I hope you can solve it! Keep in touch.

aj-ptw, 2017-09-08:
This is actually happening on both of my macs

armengol, 2017-09-08:
How's that? Did it suddenly start dropping packets?

aj-ptw, 2017-09-08:
Not sure it ever wasn't. Hmm maybe we can build an mvp of a native app that just checks sample numbers

aj-ptw, 2017-09-11:
I think this may have to do with the high sample rate of the BLE system being too fast for some mac models. Apple places restrictions on speeds and we may be feeling that. Will do more research but i say this because sending data at about half the sample rate produces hardly any dropped packets.

aj-ptw, 2017-09-12:
@armengol hey updating to the latest version of the [`noble`](https://github.com/sandeepmistry/noble).

There is an Xcode tool called `Packetlogger` that apparently dumps all the incoming bluetooth communications to the log file. I'm getting this app up and running but this should tell us if these packets are ever getting to the OS level.

armengol, 2017-09-12:
Cool, let me know how it goes!

aj-ptw, 2017-09-13:
@armengol and all

I ran the packet logger, cool app, but it showed that these dropped packets were not being registered with the mac! They never show up!! @biomurph 

[image]

I opened an issue in noble a while ago and just posted again [noble/issues/517](https://github.com/sandeepmistry/noble/issues/517)

aj-ptw, 2017-10-27:
Working to use web-bluetooth to try and solve this problem

armengol, 2017-11-07:
Great, thanks!

By the way, now that I still have to run everything in Linux, I wonder which is the easiest way to do real-time signal processing of the EEG data from ganglion. I don't care the programming language to use (Python, Matlab..). I am trying to use Lab Stream Layer to send it to Matlab but I am having issues with that. Which method would you recommend?

Thanks!

aj-ptw, 2017-11-07:
I think you should use python and look the their sendData and recieveData examples in python.

aj-ptw, 2018-03-04:
@armengol good news! I can finally close this issue with the new gui using the BLED112 driver.

https://github.com/OpenBCI/OpenBCI_GUI/releases/edit/v3.3.0-beta1

If you don't have a BLED112 dongle, you can pick one up off the web for pretty cheap and run the GUI in macOS with great quality! Finally. I've been trying to get this driver built for months now, I'm so happy macOS users can now stream perfectly!

closed 2018-03-04 by aj-ptw

armengol, 2018-04-13:
Thanks for this update!

Can I use the CSR 4.0 bluetooth dongle offered in OpenBCI website?

https://github.com/OpenBCI/OpenBCI_GUI/releases/edit/v3.3.0-beta1