OpenBCI/OpenBCI_GUI #200: Linux with CSR 4.0 dongle: Code 412Error: Please turn blue tooth on.

hack-r, 2017-08-23:
Hi,

I can see the Ganglion using `hcitool`'s `lescan` over the CSR 4.0 dongle. However, the GUI app throws the 

> Code 412Error: Please turn blue tooth on.

error. My guess is that it only uses `hci0`, which is the built-in adapter instead of `hci1` which is the dongle (?). Both the builtin adapter and the dongle are active and support BLE scanning though. I tried turning off `hci0` through `hciconfig` temporarily, but that didn't help. 

My OS is Ubuntu.

aj-ptw, 2017-08-26:
Is this problem still persisting?

hack-r, 2017-08-26:
@aj-ptw  Hi AJ. Yes, it is. Ubuntu 16.

aj-ptw, 2017-08-26:
@hack-r 

Okay, first, let's use **not** use the CSR 4.0 dongle.

Then try the GUI. This should work if your computer is BLE ready.

If this fails, then please follow the instructions found on the [`noble`](https://github.com/sandeepmistry/noble#ubuntudebianraspbian) page?

aj-ptw, 2017-08-26:
You can also try hitting the refresh list button

aj-ptw, 2017-12-13:
Pretty sure the problem is you did not use `sudo`. Closing this for now.

closed 2017-12-13 by aj-ptw

aj-ptw, 2018-04-14:
Not as far as I know! I’m not a Linux guy though, any ideas?

brylie, 2018-04-15:
Not sure, it just seems that other Bluetooth applications work without sudo permissions. It's not a big deal, so long as end users know what to expect. I only solved this issue because I found  the relevant bug report and read through the whole discussion.