OpenBCI/OpenBCI_GUI #211: Impedance values do not go over 100Hz

aj-ptw, 2017-08-26:
I set the limit to 800Hz but the frequencies never go above 100Hz.

aj-ptw, 2017-10-19:
@sunwangshu any idea where i should be looking for this?

sunwangshu, 2017-10-22:
Hello! Do you mean the limit of the FFT widget?
What do you mean by impedance value?

sunwangshu, 2017-10-22:
Do you mean the maximum frequency never go over 100Hz in the FFT widget display?
If so, this is because of the Nyquist Theorem. If you are using a sampling rate of 200Hz, then you can get 100Hz at max. Then this is not a bug.

How about when you are using higher sampling rate? Are you still getting the problem?
If so, can you do a test and print the value of fs_Hz and getSampleRate()?

sunwangshu, 2017-10-22:
The display limit of 800Hz is only the limit of the graph. It doesn't necessarily mean to set the sampling rate as 800Hz, because sampling rate is related to specific hardware.

aj-ptw, 2017-10-22:
Yes, exactly it’s when the sample rate is 1000hz and we can’t see above 100.

AJ Keller
Push The World

sunwangshu, 2017-10-23:
Could you print out the value of fs_Hz (sampling rate) for me?

I guess you are using Cyton with WIFI, then the fs_Hz (sampling rate) is defined here in BoardCyton.pde:
[image]

aj-ptw, 2017-10-24:
oh i see why all the confusion! this is the wrong issue! i meant to comment on #246 

The sample rate is 1000Hz. Let's move this over to #246

closed 2019-05-05 by retiutut