OpenBCI/OpenBCI_GUI #230: When GUI is not active, data buffer builds up... creates big lag when returning to GUI

conorrussomanno, 2017-09-04:
Try switching away from GUI, browsing the internet for a little while, returning to GUI, and testing for eye blinks in channels 1 and 2. You get a big delay between the activation of the muscles and corresponding data spikes in the time series. Interestingly enough, however, the data seems to catch back up over time.

conorrussomanno, 2017-09-04:
I just tried this on the highest data rate (1000) with the lowest latency (5ms), and the GUI was completely frozen when I switched back to it, after doing about 20 seconds of browser surfing.

aj-ptw, 2017-09-04:
hmmm that's interesting

aj-ptw, 2017-09-05:
is this still happening? Hard to duplicate on my rigs

conorrussomanno, 2017-09-05:
It is still happening. When the GUI is not the target window, it seems my computer diverts the processors to my new target application/window, and as a result, the GUI doesn't have enough processing power to keep cranking at full capacity. But when I return, it starts streaming data in double time (or maybe 1.5x time) and eventually catches back up.

conorrussomanno, 2017-09-05:
On a possibly related note, I tried introducing a 24 fps frameRate, but when active, the data doesn't stream fast enough. It builds up in the buffer (similarly to the above comment), and then I have to switch to 30/45/or 60 for it to catch back up. Does something also need to be added in dataProcessing.pde to allow for 24fps frameRate? It could be helpful for people with slower machines.. But maybe it's too slow.

aj-ptw, 2017-09-05:
i think the solution here is to do the data crunching asynchronously instead of in the update loop. so we do most of data processing when the data comes in maybe? thoughts? we could even dispatch a worker thread to crunch the new data if we do it correctly

aj-ptw, 2018-04-02:
@conorrussomanno is this still happening?

closed 2019-05-04 by retiutut