OpenBCI/OpenBCI_GUI #231: Strange cyclical noise spikes w/ WiFi Shield

conorrussomanno, 2017-09-04:
See attached screenshots. You can see there is a spike of noise once every second or so. I'm switching back and forth between Dongle and WiFi shield, and it only appears when using WiFi, so it doesn't seem to be an environmental effect.

I'm using 1000Hz and 10ms latency.

[image]
[image]

conorrussomanno, 2017-09-05:
Now this is only happening every once and a while for me. I'm not sure what is the root cause. It could be my hardware. I'm interested to see if anyone else experiences something similar using the WiFi Shield.

aj-ptw, 2017-09-05:
Are these regularly spaced? Could you try updating your firmware? Maybe something got botched the first time?

biomurph, 2017-09-05:
I don't see this problem on my wifi

aj-ptw, 2017-09-05:
could you try putting `EXT PWR` to `OFF` and use two batteries?

aj-ptw, 2017-09-05:
i think that might help with the noise

andytheengineerguy, 2017-11-27:
Hi guys, was there any progress on this?

I'm also experiencing cyclical noise spikes, which vary between 1-4hz, and only appear occasionally. The occurance of these cyclical spikes seems loosely correlated with resetting the system through the OpenBCI GUI.

I'm not using a WiFi shield, just the regular cyton EEG sensing board. It had been working well up until tonight.

I was playing around with the BIAS and SRB pins before I saw this behaviour. Any ideas?

aj-ptw, 2017-11-27:
Hi! There has been a lot of progress on this re: Cyton with WiFi shield. However this is not an issue we normally see with the Cyton over serial. Could you upload a screen shot of your spikes here?

andytheengineerguy, 2017-11-27:
Hi AJ, thanks for the quick reply! 

I actually just left work so cant take any more screenshots until I get in tomorrow morning  (different time zone here in Aus), however I did take this photo showing what happens after repeatedly connecting-disconnecting the system through the OpenBCI GUI. For some reason, the frequency of the pulses get faster each time I disconnect-reconnect, and after 6-7 iterations this is what i end up with.

Its a strange waveform thats distinctly different from random noise. While showing this waveform im unable to sense any EEG data.

Have you seen this before? Apologies if Ive gone too off track with this issue - this is really confusing me haha. Will upload pulse screenshots in the morning.

andytheengineerguy, 2017-11-27:
[image]

aj-ptw, 2018-04-02:
That looks almost like a test signal, hmm is this issue still happening?

biomurph, 2018-04-02:
Not the test signal. Some environmental noise?

retiutut, 2018-04-12:
I am experiencing this issue also when using the wifi shield with 8 channels and using two separate batteries (4xAA and 3.7v 500mah).
[image]
[image]

retiutut, 2018-04-13:
The WiFi shield firmware is v2.0.5 . Updated the Cyton firmware this morning and still having same problem:
[image]

aj-ptw, 2018-04-13:
Have you tried UDP?

retiutut, 2018-04-13:
I've mostly been using TCP. I did swap out the WiFi shield for a Daisy header and **was able to see proper EEG over BLE using the dongle.** I'll try UDP and UDPx3 later today or tomorrow.

aj-ptw, 2018-04-13:
No doubt in my mind this is 100% due to using WiFi! This can happen on some routers I’ve seen, it should magically disappear with UDP. Probably won’t happen with TCP over WiFi direct too.

retiutut, 2018-04-13:
To document: I'm using a current generation Airport Extreme mounted high on the wall. The LOS distance from the Airport to the WiFi shield was able 5 feet.

Unrelated: Had to put my iPhone in airplane mode to have uninterrupted upload for the Cyton firmware via Arduino IDE.

retiutut, 2018-04-14:
This issue is still happening over UDP, rendering the WiFi shield unusable. Was not able to find the shield using WiFi direct in the GUI.

aj-ptw, 2018-04-14:
Did you erase the network credentials?

retiutut, 2018-04-14:
I erased the network credentials again, got 10 blinks in 2 seconds on boot, and connected to the Wifi shield. The GUI still only finds this:

[image]

aj-ptw, 2018-04-14:
Weird! Which GUI version are you using.

retiutut, 2018-04-15:
Using OpenBCI_GUI 3.2.0 standalone on Mac

retiutut, 2018-04-18:
@aj-ptw What settings (hardware and software, apart from up-to-date firmware) would you recommend to resolve this issue? I was looking forward to the increased resolution over WiFi, but I get worried every time I have to change the hardware just to see if the WiFi shield works.

aj-ptw, 2018-04-18:
@retiutut could you try on a different router? If not, maybe we can do a call and I show you how WiFi direct works. Skype me at pushtheworld_aj

retiutut, 2018-04-18:
I think I have to try and update the Cyton firmware again. I'm have serious issues over BLE also :(

aj-ptw, 2018-04-20:
Hey @retiutut ! I have just finished a new gui version that will allow you to use the wifi direct for sure.

retiutut, 2018-07-01:
I have identified the cause of the problem using the Processing version! I see packet loss correlations with these visible spikes when using an Airport Extreme. I will see if this occurs over wifi direct also. Here is a screenshot that documents this issue:
[image]

engsz, 2018-12-01:
Hi guys, was there any progress on this?  I have same issue with wifi shield but not with USB dongle.

jps2000, 2018-12-01:
>>I have same issue with wifi shield but not with USB dongle.>>
So it is likely that you have electrical/ magnetical interference from the shield.
You see also in your example that despite you have 60hz notch on there is huge 60Hz in the spectrum.
This tells you what these signals are.
Do you use AC adapter or battery. The latter is important! Also for safety reasons.

You can place a grounded copper foil between shield and cyton or play around with electrode cables location.

Another idea to figure out the cause is to short the inputs. Open inputs are very sensitive.
For example I 3D printed a self-designed housing with ABS and noticed spikes appearing randomly in the time series. 
Then I noticed that just rubbing the outside of the box caused charging the plastic and hence lead to spikes.
Counteraction was to paint inner side of housing with copper laquer used for EMI 
https://www.conrad.at/de/abschirmlack-crc-kontakt-chemie-emv-35-77509-aa-200-ml-813893.html
and to connect the shield  to ground. This also improved significantly mains interference.

NOTE: Do not paint  the housing close by the antenna !!

retiutut, 2018-12-01:
> So it is likely that you have electrical/ magnetic interference from the shield.

The original post says this is likely not interference. Also, they show up on frontal lobe sensors just as strong. > I'm switching back and forth between Dongle and WiFi shield, and it only appears when using WiFi, so it doesn't seem to be an environmental effect.

paquettejp, 2019-04-26:
Could this be an artifact of the OpenBCI GUI real-time filtering when it hits discontinuous data (i.e. data packet loss)?

I am having a look at an SD card recording that for sure had these spikes during WiFi streaming to the OpenBCI GUI, but after applying a 5-50Hz bandpass and 60Hz notch, I don't seem to see these noise spikes at all.

Wondering if someone could confirm on their end. Thanks

paquettejp, 2019-05-01:
I think I may have gotten a bit closer to solving this...

Initially, I have been using a 3.7V LiPo (400 mAh) and it seemed to work, but I was getting this observed intermittent spike noise which corresponded with the D2 LED on the WiFi Shield flashing and dropped packets shown in the Processing console.

Probing the 3.3V rail (with a short spring ground), I noticed that there were pulses that caused large droops (0.2-0.6V) and very bad oscillations that could were audible. Switching out the LiPo to my adjustable DC supply, I noticed that in the range of 3.5 to 4.75V there was bad oscillation and droop. Once I supplied over 5V, the droop and oscillations went away.

With the WiFi Shield and Cyton boards connected to the OpenBCI GUI, I noticed that supply voltages below 5V caused artifacts to show up in the EEG channels and packets dropped. When I increased above 5V, I did not observe any strange artifacts or packets lost.

As for the spikes in the stream, I believe these are artifacts of the digital filters (notch and bandpass) when applied to zero-data. When the stream is started, the first packet of data is all zeros (for the EEG channels). The packet immediately following this contains actual voltage readings from each channel and will more-than-likely be at a much larger magnitude. Due to this discontinuity, the digital filter outputs generate these spike artifacts.

I could not reliably simulate the dropped packets due to the WiFi Shield acting weird, but by turning the stream ON and OFF, I was able to generate the pulses and dropped packets. I have attached an XLSX file containing side-by-side SD and stream data, noting the zero-data packets when the stream was turned ON and OFF.

[sd_stream_cyton_data_comparison.xlsx](https://github.com/OpenBCI/OpenBCI_GUI/files/3134005/sd_stream_cyton_data_comparison.xlsx)

Here is the output of applying the same digital filter settings (5-50Hz bandpass, 60Hz notch) as the OpenBCI GUI to the data collected on my SD card:
https://imgur.com/a/z5M8EPf

In my opinion, I think that this issue stems from the WiFi Shield requiring a supply with enough headroom so that it maintains a solid connection and filtering artifacts when applied to discontinuous/zero-data packets.

wjcroft, 2019-05-01:
@paquettejp thanks. I'd be interested in your thoughts on this recent forum post. We haven't verified this yet in the lab with an OpenBCI manufactured board. His oscilloscope traces were taken from a Wifi Shield, he built himself using OpenBCI board layouts. 

http://openbci.com/forum/index.php?p=/discussion/2108/wifi-shield-power-supply-resonant-circuit-noise-ringing-also-audible

paquettejp, 2019-05-01:
> @paquettejp thanks. I'd be interested in your thoughts on this recent forum post. We haven't verified this yet in the lab with an OpenBCI manufactured board. His oscilloscope traces were taken from [...]

Hi William, that was my post on the forum.

wjcroft, 2019-05-09:
**Possible workaround for Wifi Shield RF power output spikes, that are injecting interference into EEG**

The ESP8266 has an optional internal 'AT' command that can reduce the RF power output level of the transmitter. Right now I believe the shield runs at the default maximum power level. Which is probably overkill for most applications that are within 10 or 20 feet of the receiver. This high power BOTH has the potential to disrupt sensitive EEG signals, but also has caused measurable dips and resonance / ringing in the power supply circuitry. See @paquettejp (Jason Paquette) 's thread here on the forum:

http://openbci.com/forum/index.php?p=/discussion/2108/wifi-shield-power-supply-resonant-circuit-noise-ringing-also-audible

The ESP8266 commands to set the power level are documented here:

https://github.com/beckdac/ESP8266-transparent-bridge/issues/53

Page 15 and 16 of this ESP document,

https://www.itead.cc/wiki/images/5/53/Esp8266_at_instruction_set_en_v1.5.4_0.pdf

I think this is worth a try. Not sure yet if this power level should be user settable, or if we could just drop back the power level to one that would work for most everyone. And substantially reduce the noise effects. Certainly there may be some noisy environments where high power is necessary, but in those situation there may be better workarounds such as just getting closer to the receiver.

Mentioning @produceconsumerobot @andrewjaykeller

wjcroft, 2019-05-09:
From the Arduino API, the power is set using this function, ESP8266WiFiGenericClass::setOutputPower(float dBm), which calls the system_phy_set_max_tpw(val). See line 356 of:

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp

The 'AT' command I mentioned previously must be invoked by the system_phy... function.

retiutut, 2019-09-12:
[image]

wjcroft, 2019-09-12:
Is this thread relevant?

http://openbci.com/forum/index.php?p=/discussion/2108/measured-shield-regulated-3-3v-supply-has-6v-dip-at-1-second-intervals-with-resonant-ringing

On that thread, Jason (alwayswearshats), shows that shield power supply glitches can result in spikes and packet loss. Adding a capacitor as discussed in the thread, improves power supply quality markedly.

William

retiutut, 2019-09-12:
I rarely use Cyton and WiFi for all of the reasons on this thread. The noise spikes are one thing, but this packet loss is just out of hand even after all this time. I would like to see a new WiFi board!!!

Just to further document what happens, the noise spikes appear cyclically and when there is packet loss. ALSO, appears in SD card recordings after being properly converted and viewed in the GUI.

[image]

wjcroft, 2019-09-12:
Here is a 680uF cap you could try. Since the cap is 8mm in diameter, and the inter-board spacing is just a bit more than that (say 11mm), would need to be positioned to avoid other components on the board below the shield.

https://www.amazon.com/pcs-Panasonic-FR-Capacitors-680uf/dp/B072SS6YBJ/

My suggestion would be to tape the cap to the bottom of the shield, then run leads to existing large ceramic cap on the bottom of the board. Do one side at a time, so you don't disconnect the old ceramic. Enclose leads of the 680uF with some type of tubing or insulation.

wjcroft, 2019-09-14:
Had some exchanges with Jason on the thread discussing the power supply ringing. See the TI app note I posted there (and below). It exactly describes why the 100uF ceramic cap used on the bottom of the Ganglion, causes huge ringing and noise. Wish Joel or AJ had seen this before choosing that design. My guess is that the noise generated, is more disruptive to the Cyton than the Ganglion, due to the ADS1299 acting as kind of an antenna for high frequency noise pulses generated by the Shield. Components on the Ganglion are less complex and lower amplification, compared to Cyton.

http://openbci.com/forum/index.php?p=/discussion/2108/measured-shield-regulated-3-3v-supply-has-6v-dip-at-1-second-intervals-with-resonant-ringing#latest

TI app note: http://www.ti.com/lit/wp/snoa842/snoa842.pdf  "Engineers note: Capacitors are key to voltage regulator design"  

In summary, the LDO (low drop out) linear regulator as used on the Ganglion, plus the low ESR ceramic cap on the output, generate excessive ringing and noise due to the low ESR. Solution is to use a cap with both more microfarads, and more ESR. Electrolytic or tantalum looks promising. The previous 680uF electrolytic I happened to find on Amazon was an ultra low ESR; good for some applications, but not LDO linear regulator. Will reorder a more suitable cap for trial, using tips in the TI app note.

jps2000, 2019-09-15:
I can not test it but it is important where tho place the cap. As it serves as a local battery it should be as close as possible to the load. In this case when the ESP is drawing current spikes the cap should be close to its VCC and GND pin. Ideally it is a tantal 100u or so having a low ESR. The cap close to the LDO regulator should be within data sheet recommendation  ( 1-10u ).
In this case I see that the layout is poor.
I  suggest to try a cap directly from pin 8 to the ground plane ( scratch away the solder resist)
You can even try to solder it to the shield of the ESP
and to remove the 100uf  on the other side

furmanlukasz, 2019-10-02:
Hi, having the same issue over UDP but I notice that the signal goes right after disconnecting electrodes , the spikes go off, and the signal i nearly railed, so it is not the packet loose. I saw that the 'gold cup' electrodes make can bee the trouble, any time spike goes up the same happens to the impedance of electrode(just click the 'om' icon to see if it happens the same in yours setup). They just far away from the desired quality to make good measure. I will test it over Ag electrodes when they arrived so I hope It will solve the problem. Maybe somebody can test it earlier. Just a hint. But true silver electrodes cos 10 time the ' Chinese gold cup'and have a screen in cable that can be grounded also.  The screen between this two boards are also the good idea , perfectly can be grounded from one side cyton ground and the other side wifi ground.

retiutut, 2019-11-07:
I think it's time to close this issue as it exists here on the OpenBCI_GUI Repo.

Packet interpolation is live as of GUI 4.1.7-beta.0, and it resolves most packet loss spikes.

While this does improve the data received and display in the GUI, it doesn't fix the hardware problems that cause cyclical noise spikes when pairing the WiFi shield with the Cyton or Cyton+Daisy, regardless of power configuration or WiFi connection settings.

As previously indicated in the conversation above, it seems a hardware fix is required, either to existing WiFi shields or in the development of a new board/shield of some kind.

wjcroft, 2019-11-07:
Richard, I can understand why you want to close this as related to the GUI. And especially now that you have the interpolation going. But this thread documents a serious hardware issue that has yet to be solved. 

So it needs to be kept open SOMEWHERE. Perhaps by referring back to this thread from an open issue on the Wifi Shield hardware or software repo. The material in this thread is documenting some of the engineering fixes required.

This is the related thread in the other repo:

https://github.com/OpenBCI/OpenBCI_WIFI/issues/82
"WiFi Shield dropped packets"

I suggest changing the title on that thread from the original title, which was restricted to Direct mode. You might also consider changing the title on this issue thread to match.

William

Joe-Westra, 2020-01-20:
Hi @retiutut and @wjcroft,

Thanks for providing this detailed overview of the cause of the noise spikes, they were maddening when unexplained.  I am using a cyton+daisy+wifishield to transmit 16 channels at a minimum sample rate of 250hz.  Is it correct that the only option to capture this stream reliably would therefore be to use the cyton+daisy+SD card and transfer the data via BLE at session termination?

Do you have any estimation of when a fix for the wifishield (or it's next iteration) might be available?

Thank you!

wjcroft, 2020-01-21:
The shield power supply is helped, by running it on a separate AA battery pack. So TWO batteries, one on Cyton, one on Shield. Also UDPx3 is more robust than TCP, in general. Oddly some people have trouble with UDPx3, in which case fall back to TCP.

Joe-Westra, 2020-01-27:
Thanks for the feedback @wjcroft, I'll let you know how successful we are when we get a chance to try the two separate battery packs.

stellarpower, 2023-01-05:
Are we still expecting to see these sorts of issues? I don't see many references to Wifi development in the last few years. I have major regular noise artifacts in my datastream, and am going round an round in circles trying to find out what they are caused by:

[image]

Any time I think I have isolated the cause and pinned them down, they come back again. Always that distinctive shape. Is this still a known issue with the wifi mode of connection?

wjcroft, 2023-01-05:
@stellarpower you are correct. The board you are using has the same issue documented here,

http://openbci.com/forum/index.php?p=/discussion/2108/measured-shield-regulated-3-3v-supply-has-6v-dip-at-1-second-intervals-with-resonant-ringing

The Wifi Shield works perfectly with Ganglion. Which leads one to believe that the firmware interaction between the Cyton mainboard and the shield, is over-stressing the ESP8266 processor on the shield. Causing power supply spikes when the ESP is doing extensive transmissions. Thus it is 'possible' that a firmware fix on Cyton might resolve this.

However the link above explores better power supply filtering on the shield as another potential route. Though not perfect, as you can see by reading the thread.

Unfortunately OpenBCI Staff are largely devoted to Galea hardware / software development at this time. Timeline uncertain as to when a Wifi Shield fix may be implemented. I have suggested the Cyton firmware route in the past. As this would avoid any hardware mods to the Shield board. And be far superior for existing Shield owners.

I assume you are using the UDPx3 mode, and separate battery power supplies (2 AA packs). This seemed to provide the most stable results. In part because the voltage is high enough (6V) to ride out any dips.

William Croft
OpenBCI Forum admin

stellarpower, 2023-01-05:
Thanks for getting back to me so quickly.

> I assume you are using the UDPx3 mode, and separate battery power supplies (2 AA packs). This seemed to provide the most stable results. In part because the voltage is high enough (6V) to ride out any dips.

I'm using an unofficial board, and both the wifi + daisy combined shield and the main board are powered by a rechargeable Li-ion (I think) battery. My inexpensive multimeter is reading 4V across the supply rail.

According to the description, the manufacturers had improved upon some of the original issues around the noise  - I have asked and am waiting for clarification.

On the UDP mode - in the v5 GUI, this option has been removed. I believe it is being used by default, but am not sure how I can verify.

Is there any known difference when using software other than the GUI? Running it on my laptop last night, the GUI was using a huge amount of CPU and drained the (laptop) battery within about 40 minutes. I have seen [this bug](https://github.com/OpenBCI/OpenBCI_GUI/issues/349) that suggested similar artefacts ocurred when the software on the local side lagged and wasn't able to keep up with the signal. Is there any chance simply using a program in Brainflow to dump the data to e.g. CSV, and then analysing it later, would improve things at all? My next thought was to try this as using the GUI to stream the data is one variable I haven't yet removed from the equation.

So if it can be powered from 6V instead - are both boards then ambivalent to the voltage they're powered from, within reason? I would have assumed that this would need to be relatively exacting, or given that they are designed to run off batteries, do they in fact have internal voltage regulators that are more sophisticated than I would have assumed? 

In terms of diagnosing this correctly, is there anything else that spikes looking like this could likely be? Should I be seeing packet drops when this occurs in the GUI widget? As it is currently not registering any at all, having fixed a previous issue in this regard.

And if this is definitely due to wifi issues, is there anything that can be done to mitigate this? One or two channels would be sufficient for the time being, if fully powering off some features reduced the draw causing the spikes. Otherwise it looks like I've bought an expensive doorstop.

Thanks for your help.

wjcroft, 2023-01-06:
@stellarpower, thanks. Please do not post any more comments on this thread. Instead post further discussion on your Forum thread. You can quote the last comment you posted above. This thread is more devoted to the base issue with Cyton + Shield, and is not for troubleshooting specific workarounds.

https://openbci.com/forum/index.php?p=/discussion/3502/packet-loss-issue-cyton-wifi#latest

stellarpower, 2023-01-07:
I'm happy to keep the discussion there, although am yet to receive any comments, but the main reason for adding here was to confirm if this issue is still reproducable, given there's been no activity in some time. Equally for #349, I need to do some more work on the graphing, but I'm not sure I'm necessarily seeing the same artefacts when the OpenBCI GUI is removed from the equation, in which case that may want to be re-opened, or linked to a fresh issue.

If it seems from the forum post that this is in fact due to the power spikes then I will add any new relevant details here.

Cheers

philippitts, 2023-01-17:
I appreciate all the discussion here and on the forum. The WiFi shield has been pulled for a hardware update, and we don't have an ETA on its availability at this time. While OpenBCI does make some hardware designs open source, we cannot support third-party adaptations of this hardware as factors beyond our knowledge or control may have been changed. 

As mentioned by @retiutut in [this comment](https://github.com/OpenBCI/OpenBCI_GUI/issues/231#issuecomment-551256067), the fixes for this issue appear to be hardware and firmware related for official devices sold by OpenBCI. I'm closing the issue as the fixes for this problem are not related to the GUI.

closed 2023-01-17 by philippitts

stellarpower, 2023-01-24:
I don't know if you maintain open issues on hardware- and design- related threads in relevant repos, but if you do then I agree that the spikes on the power supply itself (as opposed to the fixed bug regarding packet loss, that should be seen in the packet loss widget, which I think is separate and was why this particular issue was originally opened - so the thread has become a bit tangential) would be best discussed there - and if so, a link here might be useful if anyone stumbles across this issue again and would be best forwarding comments towards ongoing discussion ;)

FWIF also appreciate it's perfectly reasonable not to be able to dedicate developer time to issues if these are limited to DIY or third-party fabrications of the board. I guess the odd thing may crop up within a third-party device that reflects on the designs or the software that would be beneficial to consider across the board, but in the general case that makes sense to me. I think trying to be inclusive of other implementations of the hardware - albeit on a second-class basis - where it seems that changes may be made that would affect all devices, and the issue hasn't been isolated just to that hardware, is more than a fair policy.

Cheers!