OpenBCI/OpenBCI_GUI #246: Cyton Wifi FFT max always at 100Hz

biomurph, 2017-09-05:
Running Wifi and Cyton. If I choose 250Hz sample rate or 1000Hz sample rate in the setup, I see the FFT only graphs up to 100Hz...

aj-ptw, 2017-09-05:
thank you!! I see the same problem too!!

aj-ptw, 2017-09-05:
any thoughts about this mr @biomurph?

biomurph, 2017-09-05:
Well, there was an issue that I brought up about not being able to select 100Hz max or 500Hz max depending on what sample rate you use (Nyquist)... So it seems like when that got fixed it might have broken this somehow?

aj-ptw, 2017-09-06:
ah okay that's a good place to look

sunwangshu, 2017-10-27:
Can you run the board with 1000Hz, then print out the value of getSampleRateSafe() and fs_Hz in the draw() loop of OpenBCI_GUI.pde?

aj-ptw, 2017-10-27:
It says 1000

sunwangshu, 2017-10-27:
OK I'll check it out later about buffer stuff

aj-ptw, 2017-10-27:
let's do a Skype!!

sunwangshu, 2017-10-28:
Fixed

aj-ptw, 2017-12-13:
fixed in v3.2.0

closed 2017-12-13 by aj-ptw