OpenBCI/OpenBCI_GUI #252: V 3.0 GUI unable to detect BLE devices

iblues93, 2017-09-07:
I am using Ganglion board with OpenBCI GUI to use the LSL protocol to stream data to Matlab R2015a for BCILab 1.4 toolbox usage.

I have downloaded V3.0 OpenBCI GUI, but the BLE devices are not detectable. The bottom tool bar displays a message "Wifi devices refreshing". Is this a bug which actually searches for devices through Wifi rather than BLE?

Anyway, I went ahead to download V2.2.1 OpenBCI GUI to test out the networking widget. I have tried to start stream on LSL but the entire GUI hangs instead. May I ask if there are any steps I have to take in order to get this to work? Thank you!

aj-ptw, 2017-09-07:
Is this on windows? Did you start the hub first? Did you use the zadig tool to swap the driver of you BLE to WinUSB?

aj-ptw, 2017-09-07:
> The bottom tool bar displays a message "Wifi devices refreshing". Is this a bug which actually searches for devices through Wifi rather than BLE?

I opened #253, this is from the BLE not starting properly i believe. You BLE driver is not starting most likely because the driver has not been overwritten with zadig or the hub was not started prior to launching the GUI. 

Thanks!

iblues93, 2017-09-07:
Hi aj! Yes, I am running on windows 7. The hub was started first, but I don't think i started it as an administrator. I will try it again when I get home. My CSR bluetooth dongle has been installed as WinUSB, so I think that theres a bug in the GUI. The V2 GUI can detect the board but not V3.  But it may have been due to the hub not running in administrator mode, so I will try it all and get back to you again.

Thank you!

aj-ptw, 2017-09-07:
@iblues93

Ah windows 7 is tougher, please check out [this forum post](http://openbci.com/forum/index.php?p=/discussion/918/ganglion-and-windows-7)

aj-ptw, 2017-09-07:
If problems continue to persist with Windows 7 you can get a wifi shield https://shop.openbci.com/collections/frontpage/products/wifi-shield?variant=44534009550

The problem stems from the fact that the Bluetooth specification the ganglion, and all ble devices, was made after windows 7, around when windows 8 came out.

iblues93, 2017-09-07:
Ahh I see. That is indeed a problem for me since its my final year project... Will it work if I switch to Linux ?

iblues93, 2017-09-07:
Or will it be alright if I run it on OpenBCI GUI V2? Apparently it can detect my bluetooth, but the problem is that the networking widget doesn't work properly. The OSC protocol is able to start stream, but on matlab, I am unable to stream it. The LSL protocol just hangs the entire gui upon start of stream... Is there a fix for this or did I install something wrongly? I am running the GUI as a standalone, should I run it through processing?

aj-ptw, 2017-09-07:
Did you drag and drop the `OpenBCIHub` to program files? What is the error code that comes up with V3? Can you confirm that you are able to use the BLE with v2? If so, that's a bug and something is up that I may be able to fix. 

You will have more success in the linux world.

The steps are

1. Download for os
2. Drag and drop `OpenBCIHUB` into Program Files
3. Drag and drop `OpenBCI_GUI` into Program Files
4. Start the hub, helps to run as admin
5. Start the GUI, helps to run as admin

iblues93, 2017-09-07:
Omg aj, I got it to work already on V3.0! I didnt run on admin privillege level yesterday therefore it didnt work. Now it can detect my board + stream into the GUI. Now I will try to use the networking widget with matlab. Thanks a lot! :)

aj-ptw, 2017-09-07:
@iblues93 woo

I'm going to close this issue and if you have trouble with the LSL, please open another issue!

aj-ptw, 2017-09-07:
I also recommend:

_Enter the new OpenBCIHub folder and right click on the executable -> properties -> compatibility -> tick “Run this program as an administrator”._

closed 2017-09-07 by aj-ptw

iblues93, 2017-09-07:
Yup, I have done that already. The GUI now works. Now the remaining problem is the LSL which just hangs the GUI when I start streaming... I will open another issue.

aj-ptw, 2017-09-07:
Okay when you open it, please tag me! Also try running from processing, as i hear that solves the LSL problem for some!