OpenBCI/OpenBCI_GUI #254: OpenBCI GUI hangs when LSL starts streaming

iblues93, 2017-09-07:
Hi guys, I have tried to stream data using the LSL on the GUI, however, the GUI hangs after I started the stream. Is there a remedy to get this to work?

-Windows 7
-OpenBCI GUI V3.0 (both hub and GUI are running with admin privilleges.
-Ganglion

iblues93, 2017-09-07:
*Update*

I have installed processing to run the GUI. Apparently, there is a problem with the LSL function to stream. Here is a screenshot of the error message. I will try  to see what I can do from here.
[image]

aj-ptw, 2017-09-07:
Awesome! I will patch this later today and publish an update! Thanks!

iblues93, 2017-09-07:
Thank you so much! Looking forward to it :)

aj-ptw, 2017-09-08:
Please pull master for the bug patch

closed 2017-09-08 by aj-ptw

iblues93, 2017-09-09:
Thank you so much :)

aj-ptw, 2017-09-11:
@iblues93 can you please test the new release to make sure this bug is really fixed?

https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.0.1

iblues93, 2017-09-12:
sorry for the late response, been busy the past few days. I will test it later when my lesson ends

iblues93, 2017-09-12:
Good day aj-ptw! I have tried it through processing and it works now. However, if it is runned as a standalone, it doesn't.

PR #255 (2017-09-08): FIX: #254 networking array overflow if not 8 chan
# v3.0.1

### Bug Fixes

* FIX: #254 LSL, UDP, OSC ArrayIndexOutOfBoundsException Stream with 4 or 16 channels
commit 979f887b (2017-09-08): FIX: #254 networking array overflow if not 8 chan
commit c2726b1f (2017-09-08): Merge pull request #255 from aj-ptw/development

FIX: #254 networking array overflow if not 8 chan
PR #256 (2017-09-08): Development
# v3.0.1

### Bug Fixes

* FIX: #254 LSL, UDP, OSC ArrayIndexOutOfBoundsException Stream with 4 or 16 channels