OpenBCI/OpenBCI_GUI #257: Inputs frequently showing RAILED, high input amplitude

brylie, 2017-10-05:
I have an OpenBCI Ganglion board, and have closely followed the Getting Started guide (even the electrode placement/colors). The impedance values for all four channels are < 30 kΩ. However, when running the data stream, there seems to be abnormally high amplitude, with the word RAILED showing up frequently in all four channels:

[image]

Is this a normal looking data stream? What can I do to improve the data stream, e.g. reducing the amplitude?

aj-ptw, 2017-10-16:
That is very strange looking data indeed!

Can you tell us what your OS is?

aj-ptw, 2017-10-16:
Also @biomurph thoughts?

brylie, 2017-10-16:
I am running Ubuntu 17.04 (Kubuntu).

aj-ptw, 2017-10-23:
Hmm this looks like dropped packets, could you share one of your saved data files?

brylie, 2017-10-23:
Here is some of the raw data saved by OpenBCI GUI:

[OpenBCI-RAW-2017-10-06_09-48-55.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/1407748/OpenBCI-RAW-2017-10-06_09-48-55.txt)

Obad94, 2018-11-14:
I'm also having the same issue . Have you got any luck to overcome this issue ?

Obad94, 2018-11-14:
@andrewjaykeller ?

daniellasry, 2019-04-19:
I believe this ganglion impedance issue was fixed by @retiutut

closed 2019-04-19 by daniellasry