OpenBCI/OpenBCI_GUI #258: Wifi Shield - Daisy streaming data @250 Hz while Cyton streaming at 1000Hz using wifi shield

msmadan, 2017-10-05:
[OpenBCI-RAW-2017-10-03_22-15-03.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/1360668/OpenBCI-RAW-2017-10-03_22-15-03.txt)

Hi @pushtheworld and @wjcroft.
I tried the wifi shield with cyton board and daisy
 for some reason 8 channels from cyton boards the data is streaming at 1000hz
but from the daisy channels I am getting data only at 250hz , I am using the GUI to record data.
Is there something i might not be doing.

When i review the data , I am  getting daisy data(16 channel data)  only after every 3 readings of the cyton data(8 channel data)
for eg

I am getting  

	line 1 - 16 channel data
line2 - 8 channel data
line3 - 8 channel data
line4 - 8 channel data
	line 5 - 16 channel data
line6 - 8 channel data
line7 - 8 channel data
line8 - 8 channel data
	line 9 - 16 channel data

I made sure the config is  set at 1000Hz

(%OpenBCI Raw EEG Data
%
%Sample Rate = 1000.0 Hz
%First Column = SampleIndex
%Last Column = Timestamp 
%Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data)

any ideas why?
raja

aj-ptw, 2017-10-05:
I am pretty sure this is coming from some issue where the daisy without wifi comes in two different packets. I've been looking into it a bit and will be able to test with a daisy board in a bit. Does this weird data happen every time you run the GUI? Even after a fresh restart?

aj-ptw, 2017-10-05:
I agree though that you are getting the daisy at 250Hz at least to the log file, this is most likely not happening on the actual hardware.

msmadan, 2017-10-05:
Yes, I get this weird data every time I use the Daisy ,even after a fresh restart

aj-ptw, 2017-10-05:
Great. I should be able to duplicate and patch this! Give me a couple days, at a conference, but this should be fixed by Tuesday.

msmadan, 2017-10-12:
Hi AJ 
Did you get a chance to patch this issue?
raja

aj-ptw, 2017-10-12:
Working on it!

It's in the firmware. Not the gui

AJ Keller
Push The World

aj-ptw, 2017-10-12:
Or a driver issue but this is getting attention. Sorry was at a hackathon past weekend

msmadan, 2017-10-16:
Just realized I am not getting accelerometer data via wifi , with Cyton and Daisy combo. Can you please let me know if I am doing anything wrong
-Raja

aj-ptw, 2017-10-16:
Hi I've been working on testing new firmware and verifying drivers for this. I am still verifying 1.2.0 wifi firmware but it's looking good. Will put this new version out next week. Can you open another issue for the no accel data? That's a bug too! Thanks so much for helping and I'm sorry these bugs are hitting and effecting you. We will get this software and hardware stack stable through relentless persistence and opening issues for every bug you find! I'm planning to really focus on the GUI this week as well and fix connectivity issues.

aj-ptw, 2017-10-16:
I don’t think you’re doing anything wrong, you’re just finding bugs! And for that I thank you!!

msmadan, 2017-10-16:
Do you want me to open an issue in wifi section or GUI section

aj-ptw, 2017-10-16:
GUI section, just like this one! And thanks!!!

aj-ptw, 2017-10-16:
I think this may be in the driver actually, not the firmware.

msmadan, 2017-10-16:
Were you able to replicate the issue I was having? If it is driver issue what might be the fix

aj-ptw, 2017-10-16:
Yes I’m duplicating. The fix is finding the bug, patching the library, and building and releasing a new hub.

aj-ptw, 2017-10-16:
At first I thought it was GUI, then firmware and now I see its driver. So it’s just a process!

msmadan, 2017-10-16:
Thanks for your help in this regard.

aj-ptw, 2017-10-16:
Crap i'm back to thinking it's firmware! I have the wifi shield sending data out of it's serial debug port and sure enough the zeroed daisy data is there!! What is really strange is that I have been running this daisy from the NodeJS WiFi driver and the daisy data is fine every time!

aj-ptw, 2017-10-16:
Okay one step further down, getting closer! This bug is present in the Cyton firmware! Wow, I'm beginning to think I should have reread the title of this issue! This seems to be happening right after the channel is set so maybe my thought now is that something is broken with the sample rate setting on the cyton firmware which is resulting in the Daisy chip only doing 250Hz and not changing!

aj-ptw, 2017-10-16:
Found it!!!!

msmadan, 2017-10-16:
Awesome.......,what was the issue
-Raja

aj-ptw, 2017-10-16:
Hi! Okay so I just released a new version of Cyton firmware, v3.1.0 which includes this bug fix, the problem was the daisy was not getting it's sample rate changed from default of 250! Please update your cyton firmware and data will come in at requested sample rate.

msmadan, 2017-10-16:
OK, I will update the firmware , get some data and let you know how it works.Thanks for the help

msmadan, 2017-10-17:
Tried the new firmware. Works as promised.
Thanks
Raja