OpenBCI/OpenBCI_GUI #260: Ganglion not detected on Mac OS

VessoVit, 2017-10-13:
Hello folks,

I been having troubles with connecting to my Ganglion board, after updating Mac OS X to 10.13. The devices doesn't get listed,  gives me an Code412 error: "turn bluetooth on" 

I tried both 2.2 and 3.0.1 version of the GUI, but no effect. Also did the tutorial part. It used to work, not anymore after update/ new machine.

aj-ptw, 2017-10-13:
I am seeing this problem too! t's from an upstream dependency! See [bleno/#341](https://github.com/sandeepmistry/bleno/issues/341)

macOS changed all the system commands for the low level driver events so none of the events are being fired. I started working on a pull request to fix these problems [bleno/#343](https://github.com/sandeepmistry/bleno/pull/343)

aj-ptw, 2017-10-13:
@VessoVit hi! 

Can you please replace the `OpenBCIHub` file with the file mac downloaded from here: https://github.com/OpenBCI/OpenBCI_Ganglion_Electron/releases/tag/v1.3.3-beta1

To find the hub, right click on the `OpenBCI_GUI` in your `Applications` folder, and select "show contents", then navigate to Contents -> Java -> data folder.

Move the old hub to your desktop or drag and drop the new hub v1.3.3 into the data folder and replace the old hub.

aj-ptw, 2017-10-13:
Hi please re download the GUI from the main downloads page. It is working with 10.13 http://openbci.com/index.php/downloads 

please make sure to run from your applications folder!

closed 2017-10-13 by aj-ptw