OpenBCI/OpenBCI_GUI #263: data streaming with wifi shield: keeps stopping after 1-2minute, have to restart

jamoonie94, 2017-10-14:
I stop and restart data stream and I can't see anything changing (time series, FFT plot etc). Does anyone else have this issue, and also a solution?

aj-ptw, 2017-10-14:
Are you using Cyton or Ganglion? What firmware is your wifi shield? And firmware of main carrier board?

jamoonie94, 2017-10-14:
I'm using a cyton and I'm using 1.1.2 for the wifishield

aj-ptw, 2017-10-14:
Did you update your Cyton firmware? Are you on Windows or Mac?

aj-ptw, 2017-10-14:
How are you powering the WiFi shield?

jamoonie94, 2017-10-14:
Im on Mac, and come to think of it when I was trying to perform a step where I had to 'upload' to default on Arduino, it didn't work. Right now, I'm powering it with four double A batteries...

aj-ptw, 2017-10-14:
Okay cool I use Mac too. Are you passing power down to the Cyton through the WiFi shield? Is the D2 led blinking when you stream data? The GUI is able to connect? If yes, screen shots?

jamoonie94, 2017-10-14:
I'm away from home at the moment, but yes, the wifishield does power the Cyton and the GUI is able to connect (although sometimes it immediately goes to 'data stream stopped')

aj-ptw, 2017-10-14:
Hi I have seen the data stream stop too when it fails to connect for no reason. Looking to resolve a lot of connection issues this week with the GUI.

aj-ptw, 2017-10-14:
Try increasing the latency to 20ms

aj-ptw, 2017-10-16:
Hi can you please try updating your firmware to 1.2.0.13, just unzip the attached file, and follow the [guide here](http://docs.openbci.com/Hardware/12-Wifi_Programming_Tutorial#wifi-shield-programming-tutorial-program-wifi-shield-over-the-air)
 
[DefaultWifiShield.1.2.0.13.bin.zip](https://github.com/OpenBCI/OpenBCI_GUI/files/1387914/DefaultWifiShield.1.2.0.13.bin.zip)

jamoonie94, 2017-10-16:
Hey aj, I've tried both and now in the GUI the firmware for the wifishield says that its 1.1.3. Even with this update and the increased latency to 20ms, the data streaming seems to stop (or freezes) after about 2 minutes.

aj-ptw, 2017-10-16:
Did you try the 1.2.0.13?

jamoonie94, 2017-10-16:
I get the following message:
Update error: ERROR[10]: Magic byte is wrong, not 0xE9

jamoonie94, 2017-10-16:
and the problem is at both 250 and 1000 hz

jamoonie94, 2017-10-16:
also, another issue is that right after I click "start system" from the GUI startup, it immediately goes to > data stream stopped, and I have to reconnect to my wifishield.

aj-ptw, 2017-10-16:
Did you unzip the 1.2.0.13 file before uploading?

> also, another issue is that right after I click "start system" from the GUI startup, it immediately goes to > data stream stopped, and I have to reconnect to my wifishield.

About to release a beta of GUI v3.0.2 that address this issue.

jamoonie94, 2017-10-16:
I've been using the dongle instead and the streaming seems to be alright. no freezing so far.

aj-ptw, 2017-10-16:
which firmware are you running on the wifi shield?

thanks for your patience

jamoonie94, 2017-10-16:
1.1.3, and no I didn't unzip it.

aj-ptw, 2017-10-16:
ok well please unzip when you upload, the project cannot be zipped.

aj-ptw, 2017-10-16:
GitHub won't let me upload a .bin file here for some reason...

jamoonie94, 2017-10-16:
ah oops. ILl give that a go

jamoonie94, 2017-10-16:
'ok it's on version 1.2.0.13 and now it is having difficulty detecting the wifi shield, and consistently fails to connect to data source.

aj-ptw, 2017-10-16:
1.2.0.13 seems to be running well for me on mac

jamoonie94, 2017-10-16:
actually, it finally worked and now I am streaming. It hasn't frozen for about 5 minutes now.

jamoonie94, 2017-10-16:
it seems to be working fine now.

aj-ptw, 2017-10-17:
@jamoonie94 awesome, i'm keeping this issue open for now, till the firmware and everything is merged

aj-ptw, 2017-10-20:
Hi @jamoonie94 how's the firmware working for you?

jamoonie94, 2017-10-20:
It keeps going straight to "Data stream stopped" before the system can even be started. The wifi sheild is properly detected, as well as the cyton + daisy and all that.

jamoonie94, 2017-10-20:
But when it does work, it still seems to be freezing.

jamoonie94, 2017-10-20:
Also shows "Failed to connect to data source..." often as well. In sum it takes a lot of quitting and restarting to get to the actual data streaming

aj-ptw, 2017-10-20:
Which GUI are you running? I just fixed these connection issues. Will be posting another build tomorrow. You’re Mac right?

jamoonie94, 2017-10-20:
Yeah, Mac, GUI v 3.0.1 2017

aj-ptw, 2017-10-20:
Download the Mac from here it fixes some of these. https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.1.0-beta1

aj-ptw, 2017-10-27:
Closed because this was fixed in [v3.1.0](https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.1.0)

closed 2017-10-27 by aj-ptw