OpenBCI/OpenBCI_GUI #265: No accelerometer data while using Cyton and Daisy combo with wifi shield

msmadan, 2017-10-16:
I am not getting any accelerometer data while using Cyton and Daisy combo with wifi shield
-Raja

aj-ptw, 2017-10-16:
Hi i've fixed this with a new hub, will be posting a new binary for this soon.

msmadan, 2017-10-17:
Thanks AJ
-Raja

msmadan, 2017-10-17:
Hi can please let me the ETA on the new binary
Thanks
Raja

aj-ptw, 2017-10-18:
@msmadan please find here:
https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.1.0-beta1

What operating system?

msmadan, 2017-10-18:
Windows 10

aj-ptw, 2017-10-18:
Ah okay I need to run home to build for windows. Binary coming soon!

aj-ptw, 2017-10-19:
@msmadan please download binary from:
https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.1.0-beta2

Thanks

AJ

msmadan, 2017-10-19:
I will test it today and let you know how it works.
Thanks for the help
raja

aj-ptw, 2017-10-19:
Great

msmadan, 2017-10-20:
I tested the system today.I am getting the accelerometer data now.I am getting the accelerometer data at 25Hz , Is there any way this can be increased

aj-ptw, 2017-10-20:
@msmadan this is possible but you’re the first person to ask AFAIK.

I would ask that you read the data sheet for the accel on the Cyton and add the code to OpenBCI_32bit_Library for others to use!

If you do not want to code this feature, please open an issue on OpenBCI_32bit_Library with as much details as you can about the feature. We will, or someone will, add the feature in.

msmadan, 2017-10-20:
I will try to code it and see if I can get it to work . If I get it to work I will upload it.
Thanks
Raja

aj-ptw, 2017-10-20:
Awesome!

aj-ptw, 2017-10-27:
This was fixed in [v3.1.0](https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.1.0)

closed 2017-10-27 by aj-ptw