OpenBCI/OpenBCI_GUI #266: Data stream freezes when saving in the BDF+ file type

GeneralFox2, 2017-10-16:
Greetings,

I want to use the BDF+ file type so I do not have to convert the data in EDFbrowser to BDF for import into EEGLAB. 

I am able to see a continuous data stream when the selected data type is OpenBCI, however when I choose "BDF+" the data stream will stop in the interface after some time.  When this happens, the interface continues to work - I can change anything on the GUI I want (Vertical Scale, the information presented in each frame, etc. etc.. The file size is relative to the amount of time before the data stream stops in the GUI.  I choose to stop the data stream, then I can STOP SYSTEM.  I then return to the GUI, where I can try again.  If I choose OpenBCI file type, then the system works without issue.

I have shut down my anti-virus, and etc.. (BitDefender).  Any suggestions?

Thanks,
Don

aj-ptw, 2017-10-16:
Could you post a screen shot of what happens when it stops? The file is written at the end of the session so be sure to hit stop session and give it a couple seconds.

AJ Keller
Push The World

aj-ptw, 2017-10-17:
Sound like you are saying the GUI freezes while recording?

The gui may hang at the end because it has to rewrite the whole file. Give it a bit!

aj-ptw, 2017-10-20:
Don. Ouch that is annoying. Will record some long sessions tomorrow and see if we can duplicate. Great write up thanks.

aj-ptw, 2017-10-20:
How long do you stream for? I'm 3 minutes in and still streaming.

aj-ptw, 2017-10-20:
sorry didn't mean to close!

aj-ptw, 2017-10-20:
I'm not able to duplicate. Any suggestions?

aj-ptw, 2017-10-20:
Please take a screen shot so i can see how many seconds have passed and such.

GeneralFox2, 2017-10-24:
Hi,  

I'm sorry you cannot duplicate this.  I'm uncertain as to which direction to take now as I am unfamiliar with the coding behind.

I have attached the images by "selecting them" below. 

[image]
[image]
[image]

[BDF File.zip](https://github.com/OpenBCI/OpenBCI_GUI/files/1409286/BDF.File.zip)

I attached one of the BDF that was created.  Perhaps that helps?

Thanks,
Don

GeneralFox2, 2017-10-24:
Sorry, forgot the usual disclaimer:
I'm using a standard HP Pavilion.  Nothing special - no additional cards.  Windows 10.   Intel i5 6400T 2GHz with 12GB RAM.  

No funky anything - latest drivers from HP - Microsoft forces all patches now.....

Always works fine with OpenBCI format.

GeneralFox2, 2017-12-26:
Hi,

Steps to resolution:
1) I am running 3.2.0
2) I noticed steps missing in the upgrade files - we are clear as to where to put the files but there is no indication of the requirement to set admin privilege  to the files when run.

New problem:  GUI freezes when file is selected for playback.  
[image]

Clicking close exits the application after waiting for a few minutes.  GUI is non-responsive.

Thanks,
Don

GeneralFox2, 2017-12-26:
Hi,

The BDF file will not play.  I just tried to open it in EDF Browser, and I received the following error:
[image]

Sorry - case is not closed!

Don

aj-ptw, 2018-04-02:
We are working on moving the BDF+ writing over to the OpenBCI Hub and using an open source community supported EDF writer, [node-edf](https://github.com/NeuroTechX/node-edf) because EDF is just too tough to maintain in this repo.

retiutut, 2019-05-15:
BDF files can be produced, but they need some work to be usable. Closing this issue.

closed 2019-05-15 by retiutut