OpenBCI/OpenBCI_GUI #270: Can't detect Ganglion by bluetooth

s86100, 2017-10-19:
Hello,

I'm using macOS Sierra version 10.12.5 and install the newest version of OpenBCI in application folder.

After turning on ganglion board, I turned on bluetooth but it can't detect ganglion board, nor does OpenBCI_GUI.

Do I miss anything?

Thank you

aj-ptw, 2017-10-19:
Hi. Did you follow the instructions in the guide?

http://docs.openbci.com/OpenBCI%20Software/01-OpenBCI_GUI#the-openbci-gui-installing-the-openbci-gui-as-a-standalone-application

Make sure the app is in your Applications folder

s86100, 2017-10-19:
Yes. The OpenBCI GUI is in the application folder

aj-ptw, 2017-10-19:
Are you sure you didn't update to 10.13? There are known issues with this #260 

Can you try something for me?

Download the mac binary for the openbci hub (formerly called ganglion electron) https://github.com/OpenBCI/OpenBCI_Ganglion_Electron/releases/tag/v1.3.2

Now navigate to your openbci gui app in the Applications folder.

Right click and hit show contents.

Go to Java

Then data.

Inside you will see OpenBCIHub, delete it, and drag and drop the recently downloaded openbci hub version 1.3.2. 

Thanks

AJ

s86100, 2017-10-20:
No. I didn't update to 10.13.
I tried two computers. Both are macOS Sierra 10.12.5. one is iMac(2012) at school; one is my mbp(late 2011).

iMac can connect to ganglion board with the software without change software's content. But when I click "start system", the software freezes.

When I use my mbp, it just can't detect the device anyway.

I tried with my friends mbp which uses El Capitan. And it works.

aj-ptw, 2017-10-20:
Did you try the other hub?

s86100, 2017-10-20:
Yes. All tried.

aj-ptw, 2017-10-20:
The Hub 1.3.2 didn't work on 12.5? Sound like something i'm missing something.

aj-ptw, 2017-10-20:
Can you try an older version of the GUI? Try v2.2.1? http://openbci.com/index.php/software

aj-ptw, 2017-10-20:
@biomurph any ideas?

s86100, 2017-10-20:
Yes. I doubted that my laptop's bluetooth has problem, but it can connect to other bluetooth devices functionally.

s86100, 2017-10-20:
I tried v2.2.1 on both computers. It doesn't work.

aj-ptw, 2017-10-20:
This is bluetooth low energy, which is different then bluetooth classic, which audio devices such as speakers use. ble is newer 

Is there an error in the bottom of the GUI such as 412?

s86100, 2017-10-20:
No. When I use iMac, it can connect to ganglion board. But when I click start system, nothing happens, and I can't do anything except quit the program.

aj-ptw, 2017-10-20:
Try running the GUI from Processing. 

http://docs.openbci.com/OpenBCI%20Software/01-OpenBCI_GUI#the-openbci-gui-running-the-openbci-gui-from-the-processing-ide

The error messages will be there.

s86100, 2017-10-20:
ok. It works when I use processing to connect it.

s86100, 2017-10-20:
This is the error messages i got: https://imgur.com/a/req1u

aj-ptw, 2017-10-20:
I'm lost where you're at right now, what IDE are those error messages from? They look like warnings not errors.

s86100, 2017-10-20:
I'm using Processing to run OpenBCI. I successfully start the system with connecting ganglion board and run openbci hub.

aj-ptw, 2017-10-20:
can you please take a screen shot of your applications folder so i can see how the OpenBCI_GUI is placed.

s86100, 2017-10-20:
https://imgur.com/a/arHz9
https://imgur.com/a/zgBXw

aj-ptw, 2017-10-20:
@s86100 thanks so much for your continued support. While, i'm happy you are able to run from processing, I'm confused as to why the heck it didn't run on your computer as a built binary. I wonder if by chance a pop up came up that said, do you want to run this app to the hub on it's first launch and maybe you hit no? I'm really at a loss for why this is not working!

s86100, 2017-10-20:
It didn't pop up any when first launch. is this the reason causes the problem?

aj-ptw, 2017-10-23:
i don't think the pop-up not launching would cause an issue. Normally i see this when I don't put the application in the correct folder.

Zigzagzen7, 2017-11-11:
I am having issues with the new GUI as well: I can't connect to my ganglion (nor my Cyton for that matter) with OpenBCI_GUI v 3.1.0 on MacOS 10.13.  GUI is in Applications folder.  The CRS 4.0 dongle is not plugged in.  Ganglion and batteries are working fine as I can connect on a PC with an older version (2.2.0).  The GUI simply does not see any device.  What to do?

jooray, 2017-11-15:
Same setup as @Zigzagzen7 - macOS 10.13.1, OpenBCI_GUI 3.1.0 (copied to and ran from /Applications folder). Also tried the Processing GUI, same result, cannot see any device (actually, I am using the Processing sketch first, I tried with the latest version of the GUI app). Also upgraded the hub app to the latest version in the data directory, no change.

aj-ptw, 2017-11-15:
Hi we are verifying the new web Bluetooth driver for the ganglion and will integrate it into the hub! We are aware of this problem and working hard to find a stable solution!

aj-ptw, 2017-11-15:
We found out that we can use web Bluetooth inside the hub and have proved that you can still find the ganglion in chrome on Mac 10.13.

I had to redo the utility module to work in the browser and now we are able to use the ganglion in chrome on 10.13 using [ganglion-ble](https://github.com/alexcastillo/ganglion-ble) 

I flew to Cali with one of the main purposes of verifying this new driver.

lmonson, 2017-11-19:
I'm also unable to connect - macOS 10.13.1, OpenBCI_GUI latest run from Applications.  I'm happy to verify any proposed fixes!

aj-ptw, 2017-11-29:
@s86100 @Zigzagzen7 @jooray @lmonson 

Please try https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.2.0-beta3

Thanks! I'm trying to build a special version of the hub with support for 10.13!

aj-ptw, 2017-12-05:
I finally found the problem and verified on macOS 10.13! Building a new gui for y'all with the hub!

aj-ptw, 2017-12-13:
This problem is solved in the [latest GUI release v3.2.0](https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v3.2.0).

closed 2017-12-13 by aj-ptw